EdgeTX/edgetx #129: Switch layout documentation

lshems, 2021-05-30:
Did someone perhaps already make a Hardware Layout document? 
A Table of all available sticks, trims, pots, sliders and switches per radio type?

Should we differentiate between pots and sliders?

I'm about to start one, but if there is already something ...

And where would we document that?

raphaelcoeffic, 2021-05-30:
@lshems documentation is dearly needed, that's right. That can be placed into the Wiki after review.

JimB40, 2021-05-30:
Someone did and put link on rcgroups long time ago 😀

https://docs.google.com/spreadsheets/d/1jGSgQBNOv3DiY4V5XwVPbfLer7l_M4hivX_V9Kb5iZ0/edit#gid=1717622432

raphaelcoeffic, 2021-05-30:
This is impressive! We must have this in the docs to reference things!

JimB40, 2021-05-30:
Can I  share spreadshet  on git not loosing edit functinality?

raphaelcoeffic, 2021-05-30:
> Can I share spreadshet on git not loosing edit functinality?

probably not, at least I don't know how. You would basically just export a copy stored in some format, or just link to it from some wiki / readme page.

lshems, 2021-06-10:
Can we put the link in the documentation somewhere and close the issue?
Where is the documentation? Happy to add if I know where.

Trying to close clutter issues that don't have development effort linked to it.

raphaelcoeffic, 2021-06-10:
> Seems there is no easy way to publish to wiki (or ask to pull changes). Wiki is a place it should go.

Maybe in a non-editable form? Like a huge print, or something like that?

phileville, 2021-06-24:
Added this file to the website repo (https://github.com/EdgeTX/edgetx.github.io/tree/master/files) - from there we can add it to any wiki or webage for future reference.  If someone wants to make changes - they can like any other file. It can also just be simply downloaded , by right clicking and save as... https://github.com/EdgeTX/edgetx.github.io/blob/master/files/OpenTX%202.3%20LUA%20DEV.xlsx
Is that alright for everyone?

jrwieland, 2021-06-24:
just a thought, what about going back to google docs for the documentation (initial write-up) all collaborations are presented IRT and automatically saved to a single document.  OTX started with this format back in 2012 but they exceeded the server capacity and moved to github.  Once the initial write-ups are complete the file can be transferred to the wiki or saved in any format that could be accessible via web or download.

my 2 cents, but I am here to get the documentation online... like everyone else. 
also, we could also ask for another tab under discord development for conversations on this subject since it will have man questions and the need for communication between the writers.