EdgeTX/edgetx #132: Forward data from lua widget-script to lua mixer-script

wimalopaan, 2021-05-31:
*Before suggesting a new feature here, please search the open issue if there is already a request for the feature*

**Is your feature request related to a problem? Please describe.**

There is no way to forward any data from a widget to a mixer script. One can use GC or fake telemetry-data, but that is far from optimal.

There are lua-widgets that display and generate some sort of data/values, that should be transferred over RF to the model. So there must be a safe way to forward that data from an (uncritical) widget to a (critical) mixer-script.  

**Describe the solution you'd like**

There should be a set of forwarding / gateway data-structure or simple forward-variables

**Describe alternatives you've considered**

mixer- and widget script use the same lua-context. This may impose a negative affect on a mixer-script.

lshems, 2021-05-31:
There needs to be a thorough analasis done in the entire lua setup.

Ideally, all instances would run in there own sandboxed environment, and get a table passed in the environment to share data from the 'radio' environment.

Than every scripter is sure his script will work, independent of other scripts by other users.

If you want to share data, then you can, but at own risk.

Currently, with the sharing of globals, I can ruin any script that uses globals. Also, the programmer must be really careful on his scope of variables 

With every script getting his own environment, this is not an issue anymore.

raphaelcoeffic, 2021-05-31:
> Currently, with the sharing of globals, I can ruin any script that uses globals. Also, the programmer must be really careful on his scope of variables

We can probably tackle this with prefixing, I guess. But I understand your concerns.

lshems, 2021-05-31:
I would really suggest to look at standard solutions as foreseen in LUA. scoping of functions to withing their own environment is very common and well described

raphaelcoeffic, 2021-05-31:
Something that needs to be looked at as well:
- mixer is executed in a different thread
- standalone scripts should stay "standalone", I think.
- Widgets & Apps (widgets in fullscreen mode) run in the UI thread.

So, we need to be careful with locking and everything that goes with it.

lshems, 2021-05-31:
Also, their seems to be a misunderstanding about the chunks as we are using now.

There really is no need to return functions as such. Every script IS a function.

The arguments in the call to the chunk are available as ...,   and returns are not obligatory.

raphaelcoeffic, 2021-05-31:
> There really is no need to return functions as such. Every script IS a function.

Off topic ;-)

lshems, 2021-05-31:
Not really ;)

Interaction can be done with environments, or with passthrough as with
event to the run script that is called from the radio.

The way you interact depends on how you interact.

raphaelcoeffic, 2021-05-31:
>There is no way to forward any data from a widget to a mixer script

This is the topic of this issue. It's written on top.

wimalopaan, 2021-07-13:
I like to suggest a set (say 16) of simple "widget-interface-variables" like the set of global variables. 

These should not be visible to the UI and should not be accessible as sources to anything. To access them there should only be lua function like "setInterfaceValue(id, value)" and "getInterfaceValue(id)".

wimalopaan, 2021-09-24:
I would suggest to introduce a set of 16 Shared-memory-variables: ShmVars as Shm1 to Shm16. These should be simple physical int16_t and logical 11-bit w/ offset as all other values are.

Sould be accessible as setShmVar(id, value) and getShmVar(id) and should be selectable as source (Shm1 to Shm16).

jfrickmann, 2021-11-09:
@wimalopaan, please test out #817 and let me know if it works for you 😎

wimalopaan, 2021-11-10:
Thanks!
Looks good to me.
But I would like to change it also for non-colorlcd radios. I don't think that the 16 int's hurt so much for the smaller radios.

jfrickmann, 2021-11-11:
The X9D-non-2019 models are very memory constrained as they have the 4bit x 212 x 64 display and only 128 KB. 
I also do not see a usage case for radios w/o widgets?

wimalopaan, 2021-11-11:
Yes, you' re totally right! 
Fool me forgot that telemetry- and mixer-scripts on bw-radios run in the same lua-context.
Thanks!
So it is ready to be merged ;-)

closed 2021-11-13 by pfeerick