EdgeTX/edgetx #133: Checkable preflight-checklist

wimalopaan, 2021-05-31:
**Is your feature request related to a problem? Please describe.**

There should be a really checkable checklist: only iff all options are checked by user the model gets started.

**Describe the solution you'd like**

A textfile containing "[ ]" should be checkable in the above way, a file without "[ ]" should behave as normal.

**Describe alternatives you've considered**

none

JimB40, 2021-05-31:
1. Seems like LUA level functionality
2. 'Model start' is configurable depending on what you control. So very broad and not easy to define condition

ItsMeAubey, 2021-05-31:
1. would require that there is some way to launch a lua script at boot / on model load, which does not exist. There is also no way to lock out all sticks etc via lua as far as i am aware. 
2. when a model is loaded (radio boots, model manually selected), check to see if there is a checklist assigned to that model. If there is, it is run. When all checkboxes are filled, the actual model is loaded.

JimB40, 2021-05-31:
And when you want just to configure model? Need to click all checkboxes to see and change logic switch?
Not very convinced to block anything because of check list. It's like having a car that wont start until you fasten belts. Most of them just beeps.
Having this checkbox list sounds good. Preventing anything not.

ItsMeAubey, 2021-05-31:
Make a key to hold down for 5 seconds to skip the checklist, or add an option for a skip button.

pfeerick, 2021-06-01:
> And when you want just to configure model? Need to click all checkboxes to see and change logic switch?
> Not very convinced to block anything because of check list. It's like having a car that wont [...]

Keep in mind it's *you*, the end user who opts for the pre-flight checklist to be blocking by adding the checkboxes, so it's up to you whether it's blocking or not. The current OTX 2.3 implementation is to simply show the model notes text file when the model loads when the preflight checklist box is selected...

I would envisage this as making it so the checklist 
1. has tickable boxes 
2. failing to tick all of those boxes would not enable RF output for the model until they are all ticked
3. having a something like long press of enter / long press of screen to bringing up option skip the checklist

To have anything more complicated would necessitate pre-flight checklist being deeper intertwined into the menus and RF output state, and I don't see it being a feature that would be used enough to warrant that sort of work.

JimB40, 2021-06-01:
If this is an option it may do whatever you want.
Btw I'm very curious how to check flaps if RF if off. :) 
Surely for racing quad pilot or seasoned free-styler this will be first thing to switch off.

ItsMeAubey, 2021-06-01:
Nobody said it should be enabled by default...

wimalopaan, 2021-06-01:
> I would envisage this as making it so the checklist
> 
> 1. has tickable boxes
> 2. failing to tick all of those boxes would not enable RF output for the model until they are all ticked
> 3. having a something like long press of enter / long press of screen to bringing up option skip the checklist
> 
> And if you don't want the beeps and bells, don't make it an interactive checklist in the first place! ;)

That's excactly what I meant.

pfeerick, 2021-06-02:
> Surely for racing quad pilot or seasoned free-styler this will be first thing to switch off.

It's currently off by default, and there's no suggestion of it changing! 😛 And as a racing quad pilot, I can say for myself that I don't use a checklist on the TX, I'm too lazy, and do the checks automatically anyway. 

Because of safety first principles (no RF until complete), it would not be possible to do thinks like 'check flaps' prior to finishing the checklist... it would have to be prompt to do so after completing the interactive portion of the checklist is complete... Otherwise, you really are going into the territory of a complex multi-stage pre-flight lua script... which is way beyond the scope of this request ;)

JimB40, 2021-06-02:
That's why i'm opting on OS level to give a tool for creating user scenario solution instad of defining one. :)

pfeerick, 2021-06-02:
I suppose we'd have to wait till post 2.4.0 to see how changes to the LUA API progress? As for sure, being to instead say run a LUA script rather than repurpose the txt file would be much more flexible... i.e. just spitballing here... you could the first part checklist disable RF entirely / have no channel output, and then potentially have individual channels enabled and disabled as you test surfaces, and then either enable all or none depending on the status of the checklist. Or is that going too far? 😁

raphaelcoeffic, 2021-06-02:
Having the option to have an auto start LUA script might be a good solution. For the issue here, we might need a whole lot more than that to achieve this goal.

JimB40, 2021-06-02:
I'm beliver. You give a tool like model.switchRF() and you will be surprised with briliant solutions guys will come up.
100% percent sure also one of our contributors can provide default checklist.lua to be included with SD card distro.

raphaelcoeffic, 2021-06-02:
> I'm beliver. You give a tool like model.switchRF() and you will be surprised with briliant solutions guys will come up.
> 100% percent sure also one of our contributors can provide default checklist.lua to be included with SD card distro.

That sounds like a good solution to me. So we need:
- auto start LUA (with access to LCD, similar to standalone scripts)
- Control over modules ON/OFF? I think RF switch is basically just module ON/OFF.

pfeerick, 2021-06-02:
That sounds perfect... With zero knowledge of the existing LUA API, for completeness I would suggest you would need some way to get the configured module state (which also gives access to selected protocol as a bonus extra) so you can do an (manged code example) ... 

``` 
if radio.getInternalRF() != PROTO_OFF; radio.switchInternal(False);
if radio.getExternalRF() != PROTO_OFF; radio.switchExternal(False);
```

... meaning you have granular control (which could feed into the checklist as well... model link... telemetry link?) as well as only turning on or off what you need, rather than everything. 

I'll shut up now before I make a complete fool of myself talking about stuff I don't know about! 😝 Really should go look at the opentx lua ... *someday soon*...

lshems, 2021-06-02:
I already made a simple text file editor in LUA. Can already be used to add simple notes to the text file.

It needs proper review, but I'll see if I can share it as opensource. Might add to the userability. 

Especially if we get a keyboard popup

wimalopaan, 2021-06-02:
If we go the lua-way for this instead of built-in feature, then it should be useable as a built-in feature. That is: enababling a checkable preflight list should a one-tick-activateable feature. No need to install some extra lua-scripts.

raphaelcoeffic, 2021-06-02:
@wimalopaan then it might take longer until you get it ;-) having a mandatory sd card image installed should be avoided. It is not the case yet, but will be very soon.

lshems, 2021-06-02:
No sdcard necessary also for colour?

raphaelcoeffic, 2021-06-02:
> No sdcard necessary also for colour?

At least not with a pre-defined set of files. This what PR #15 is about. But you will need a SD card for storing your models and radio settings. This will be the case as well with B/W radios once YAML has been introduced.

lshems, 2021-06-02:
Well, the you have to decide if this is a 'nosdcard' function. Voice announcements certainly not and you could argue the same for this. But I leave it up to you to decide.

raphaelcoeffic, 2021-06-02:
> Well, the you have to decide if this is a 'nosdcard' function. Voice announcements certainly not and you could argue the same for this. But I leave it up to you to decide.

This does not include the voice files, that's right. They can be fetched separately if wished, just like additional themes, custom background & splash image. But we should be able to start without all this.

lshems, 2021-06-02:
Not clear. If this should work without Sdcard, lua is no option. Or I'm missing something

JimB40, 2021-06-03:
Nope you're right @lshems, Still there may be OS in-build simple solution and once OS detects sd-card option will be shown to switch to checklist.lua for more fancy or customized solution.

piotrva, 2022-01-27:
@wimalopaan #1417 should satisfy your enhancement request.

wimalopaan, 2022-01-28:
I posted my comments in #1417

PR #3564 (2023-08-29): Interactive/progressive checklist
rebased, fixed and LVGL converted replacement for #1417 

copied from #1417:
Summary of changes:
Adds an option to display check-boxes at the beginning of each checklist / model notes line. 
- Item to be currently processed has selected (active) check-box, confirmed item - selected checkbox, unconfirmed item - empty.
- You can only advance by one in the checklist (by clicking ENTER - rotary encoder click).
- You can normally scroll the checklist if it exceeds screen size and in addition it is automatically scrolled when currently processed item is at the botto [... cut]
closed 2023-08-29 by pfeerick