EdgeTX/edgetx #134: Add direct interface from lua to output-stream

wimalopaan, 2021-05-31:
**Is your feature request related to a problem? Please describe.**

Add a possibility to write directly to the output-buffer, that transferred via RF pakets. Also add a transactional / double buffered semantic, so that ist possible to prepare some data e.g. in different output channel values via non-atomar lua writes. 

**Describe the solution you'd like**

Could imagine a interface like:

- lua.prepareBuffer(index1, index2, ...) (begin transaction) (index1, ...: the channels, that should be overwritten)
- lua.setBufferValue(index, value)
- lua.sendBuffer() (end transaction)
- lua.releaseBuffer(...) dismiss the overwrite of channel values

This could be useful to transfer values that do not fit into a single channel value, but could be transfered in multiple channels. This needs an atomar possibility to set the values.

raphaelcoeffic, 2021-05-31:
> transferred via RF pakets

What do you mean exactly here? Do you mean to access the serial connection which leads to the module? EdgeTX does not send any RF packets itself.

wimalopaan, 2021-05-31:
I mean the sbus-like data-stream that is send to the RF-module.

imagine a widget that codes the state of several "virtual switches" (of a menu coded as lua widget) into the value of a channel. Due to the rounding in several places from mix to the RX-output one can use only 9-bits of the 11-bits in the values. This is a shortcoming for modelists that build ship-models or other kind of models that needs a lot of switches to turn on small motors, LEDs, sound effects and so on.
One possibility to extend the usable 9-bits (6-bits for the function (64) and 2-bits for the function state and 1-bit for config/control-messages) to more ist to combine two channels. 
But if one uses two lua.setValue() calls it is not guaranteed that the two calls are made atomic. This leads to problems since the RX may then get and output one new channel-value and one old channel-value.

I know that this is very specific to this use-case but most RF-protocols miss the possibility to transport "virtual switch" states. The only one I am aware of is SumDV3 / Hott. Additionaly the interface to the multi-module lacks this feature.

raphaelcoeffic, 2021-05-31:
I see now, so basically a way to set channel values directly without any interaction from the mixer, right? This should be possible, and fits with the idea of using a new LUA touch api for exactly this: virtual buttons and knobs. Could be interesting for everyone.

wimalopaan, 2021-05-31:
Is there any info about the api for lua virtual buttons and knobs?

raphaelcoeffic, 2021-05-31:
> Is there any info about the api for lua virtual buttons and knobs?

Yes, it’s “to be defined” :-D