EdgeTX/edgetx #135: EdgeTX startup sound does not come first

rotorman, 2021-05-31:
Tested f9aeebca on RM TX16S with Yaapu LUA telemetry widget. In my opinion, on power-up of the radio, the "Welcome to EdgeTX" sound should come first before other sounds. The following video shows, that e.g. in case Yaapu LUA telemetry widget is active, it's welcome sound "Yaapu telemetry ready" precedes EdgeTX welcome sound:
https://www.youtube.com/watch?v=Gk1ZFmFnxsc

raphaelcoeffic, 2021-05-31:
Aha! So that would be mean to force the sound queue to play the other sounds at the end. I think the audio is already capable of things like that. So it needs to be used properly, and that should work.

pfeerick, 2021-08-28:
This is one _annoying_ issue... 

This is happening because 
https://github.com/EdgeTX/edgetx/blob/5d953c85acc0ad12c450abefd35744bd62e94281/radio/src/opentx.cpp#L1819

is run before the audio queue is started at 

https://github.com/EdgeTX/edgetx/blob/5d953c85acc0ad12c450abefd35744bd62e94281/radio/src/opentx.cpp#L1861

and even though part of the AudioQueue startup task is to play the welcome audio

https://github.com/EdgeTX/edgetx/blob/8becbfcb499d919b6ba537de06778ab296a734e8/radio/src/audio.cpp#L542-L544

it seems Lua manages to get in first, and it's a "dumb" FIFO queue, and doesn't seem to have facilty for 'jumping the queue'. 😢 

It also seems that the jobs can't be re-ordered - i.e. so that LUA widgets start later.

pfeerick, 2021-11-24:
Interestingly enough, this is actually the behaviour of OTX ;)

closed 2026-02-18 by pfeerick