EdgeTX/edgetx #138: Please add HARDWARE_TOUCH option for T18/TX18S build

Ender1967, 2021-05-31:
*Before suggesting a new feature here, please search the open issue if there is already a request for the feature*

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Ender1967, 2021-05-31:
set(HARDWARE_TOUCH YES)
T18 and T1X18S have hardware touch interface, some models just don't have the touch layer on top of the LCD screen installed.
RadioKing TX18S has the same hardware as Jumper T18 and can use the same image and had the touch screen installed.
Please consider adding the hardware touch option for T18/TX18S.

raphaelcoeffic, 2021-06-02:
@Ender1967 ATM, the GT911 is not configured the right way. Feel free to use the binaries in the PR to check what I mean. We will have to wait for Jumper (or somebody else) to come up with the proper configuration, as I have other bugs to fix first.

commit 7789f554 (2021-06-15): Added touch support for T18

Resolves #138
closed 2021-06-15 by raphaelcoeffic