EdgeTX/edgetx #141: Bug: Disconnecting USB when in Storage Mode triggers emergency mode

pfeerick, 2021-06-01:
TX16S, https://github.com/EdgeTX/edgetx/commit/267375cda8f9b752674697ad7e4188886b2df5c9

I initially thought this was because I just added the betaflight LUA scripts, but it also happened when I:
1. Connect USB to computer
2. Select USB storage mode
3. (On Windows 10) safely remove the drive
4. Unplug the USB lead
5. Transmitter screen goes black
6. Emergency mode message appears
7. Press and hold power button to power off

raphaelcoeffic, 2021-06-02:
@pfeerick hmm... I had fixed that one a couple of days ago, seems it is coming back to haunt us.

pfeerick, 2021-06-02:
I think it's a new variation of that... but it was a couple days ago! 😁

raphaelcoeffic, 2021-06-06:
@pfeerick I cannot reproduce any more with the current HEAD of 2.4 (14f0aa6884f7d9), so I'm closing it now. Please re-open if that does not work for you.

pfeerick, 2021-06-06:
I'm afraid I can still reproduce this - TX16S, 'normal' USB SD mode (i.e. not bootloader USB SD mode) - on Linux, mount, eject - emergency mode. This was with the auto-switch/auto-source branch just now. For now, lets leave this closed till it can be reproduced by someone else...

rotorman, 2021-06-06:
Wanted to test this, but the 2.4 head does not compile unfortunately anymore. Fails for api_colorlcd.cpp:
```
[ 54%] Building CXX object radio/src/CMakeFiles/firmware.dir/lua/api_colorlcd.cpp.obj
C:\edgetx\edgetx\radio\src\lua\api_colorlcd.cpp: In function 'int luaLcdDrawFilledTriangle(lua_State*)':
C:\edgetx\edgetx\radio\src\lua\api_colorlcd.cpp:863:17: error: 'class BitmapBuffer' has no member named 'drawFilledTriangle'; did you mean 'drawFilledCircle'?
  863 |   luaLcdBuffer->drawFilledTriangle(x1, y1, x2, y2, x3, y3, flags);
      |                 ^~~~~~~~~~~~~~~~~~
      |                 drawFilledCircle
mingw32-make[3]: *** [radio\src\CMakeFiles\firmware.dir\build.make:1251: radio/src/CMakeFiles/firmware.dir/lua/api_colorlcd.cpp.obj] Error 1
mingw32-make[3]: *** Waiting for unfinished jobs....
[ 54%] Building C object radio/src/CMakeFiles/firmware.dir/thirdparty/Lua/src/lapi.c.obj
mingw32-make[3]: Leaving directory 'C:/edgetx/build-edgetx'
mingw32-make[2]: *** [CMakeFiles\Makefile2:409: radio/src/CMakeFiles/firmware.dir/all] Error 2
mingw32-make[2]: Leaving directory 'C:/edgetx/build-edgetx'
mingw32-make[1]: *** [CMakeFiles\Makefile2:416: radio/src/CMakeFiles/firmware.dir/rule] Error 2
mingw32-make[1]: Leaving directory 'C:/edgetx/build-edgetx'
mingw32-make: *** [Makefile:175: firmware] Error 2
mingw32-make: Leaving directory 'C:/edgetx/build-edgetx'
```
Likely not related to this PR though, but another.

rotorman, 2021-06-06:
Tested d1aac698 on RM TX16S. According to steps from @pfeerick :

1.     Connected USB to computer
2.     Selected "USB Storage (SD)"
3.     Full screen USB black and white logo is shown on radio display
4.     (On Windows 10) safely remove the drive
5.     Unplug the USB lead
6.     Power light goes off, transmitter screen goes black
7.     Power light comes back on, emergency mode message appears
8.     Press and hold power button to power off

Indeed EM. I did this while holding the DFU button of my "custom" TX16S to make it behave as stock TX16S as well - same result - EM.

rotorman, 2021-06-06:
I let it run with Segger Ozone connected via SWD and I get HardFault/STKALIGN as soon as I pull the USB plug from radio:
[image]

rotorman, 2021-06-06:
When disconnecting the USB cable, the execution goes to main.cpp _handleUsbConnection()_.
There, it calls _opentxResume()_ that calls _storageReadAll()_ of sdcard_common.cpp.
It then calls _loadModel()_ in the same file, where the execution reaches _preModelLoad()_ from storage_common.cpp:43.
This calls **deleteCustomScreens()** from layout.cpp that seems to go awry (while full screen black and white USB logo screen is displayed).

rotorman, 2021-06-06:
Thx to @raphaelcoeffic for the tip!
https://github.com/EdgeTX/edgetx/blob/d709b69a3cb5bfa8c7d09c6641bcfe83c6636882/radio/src/opentx.cpp#L1517-L1519

commit d709b69a (2021-06-06): resolves #141
closed 2021-06-06 by rotorman