EdgeTX/edgetx #144: Global function unable to select S1,S2 on Radiomaster TX16S

paulkidd68, 2021-06-01:
I was trying to set up global functions for brightness and volume using S1 and S2 in the system menu but there is no option to use them in global functions

pfeerick, 2021-06-02:
Possibly you're setting it up wrong... just like with OTX 2.3 it's a little odd because the pot is setting the value, not the state/switch ;) It's working for me on the TX16S with S1 for backlight, S2 for volume. 
```
Switch: ON
Func: Backlight
Value: S1
Enable: tick
```
And same for volume, but Func: Volume

raphaelcoeffic, 2021-06-02:
@pfeerick having a knob for Volume and/or Brightness is a pretty basic feature, why not having this as separate menu item in Radio Settings instead of these global functions? That would enable to use these knobs right after boot, which is not the case right now (Volume issue), as Global functions are enable pretty late, once the radio is fully booted and ready to operate. Many people complained that the Volume knob would be applied for the "Welcome to OpenTX" announcement.

pfeerick, 2021-06-02:
I 100% agree... I would like to see it as a main option rather than having to add via global funcctions... i.e. either a fixed value, or linked to a pot source... For both the reasons you mentioned! :) 2.4.1? ;) Would obviously have to be radio dependent... ie. T-LITE doesn't have pots ;)

raphaelcoeffic, 2021-06-02:
> I 100% agree... I would like to see it as a main option rather than having to add via global funcctions... i.e. either a fixed value, or linked to a pot source... For both the reasons you mentioned! :) 2.4.1? ;) Would obviously have to be radio dependent... ie. T-LITE doesn't have pots ;)

Well, that is unfortunate ;-) but you might want to re-purpose trims, as you probably don’t use them as a quad pilot ;-)

raphaelcoeffic, 2021-06-02:
For all these funky now options we are planning, it is very important to finish the YAML settings first, as it would enable much more flexibility with the data model. Hopefully I can finish that soon....

pfeerick, 2021-06-02:
> Well, that is unfortunate ;-) but you might want to re-purpose trims, as you probably don’t use them as a quad pilot ;-)

Wah... wah ...wahhhhhh...  but I do! :-P Just leaving the hint that it probably isn't a good option to have it say S1 on a transmitter that doesn't have an S1... ;)

eshifri, 2021-06-03:
In 2.3.11 S1 and S2 are available for GF (at least for TX16S). I'm using them exactly for Brightness/Volume.

pfeerick, 2021-06-03:
@eshifri They are available in EdgeTX (and probably OTX 2.4), and per my initial reply to the OP (second post in this issue) using S1 and S2 for volume and brightness do work. Raphael is suggesting that it should actually be a setting in the radio options (rather than being implemented as a global function), thus also allowing it to take effect almost immediately on power on, rather than with a delay. 😉

raphaelcoeffic, 2021-06-04:
@paulkidd68 Did you try to configure it as pointed out by @pfeerick ? If that works for you, the issue should be closed.

closed 2021-06-07 by raphaelcoeffic