EdgeTX/edgetx #146: EdgeTX UX/UI

JimB40, 2021-06-02:
Hello,

There are few quite interesting high-level concepts regarding "how EgdeTX should work in terms of UX?". They pop-up here and there in a form of issue or pull comments or discord discussions. So decided to make document we can relate to.

I'm not fan of sharing 'work in progress' as this sometimes creates too many 'low-level' questions still it seems this be open source projects core. To throw initial idea and let it be defined in this social pot. If you perceive this as philosophical bull-shit no problem :I often can't understand what you talk about on deep coding level :). Fundamental change in smartphones UX started with controversial and bold statement 'It should have one button'. On top of that we talk about changes in 2.4.x or later. 

Here is first EdgeTX UX/UI Concept draft based on what I've read & heard here plus few my insights.
Don't jump on me with 'Exactly how it should be done?' yet. 
I'm working on point 6 which is UI Concept or as @raphaelcoeffic called it 'story board' :)

Anyway throwing thing into the pot.

Cheers.

[EdgeTX_UX-UI_rev1.pdf](https://github.com/EdgeTX/edgetx/files/6582264/EdgeTX_UX-UI_rev1.pdf)

raphaelcoeffic, 2021-06-02:
@JimB40 many good ideas here. This needs to be refined and broken down.

pfeerick, 2021-06-06:
Some really good points there, and I think this is exactly the direction that OTX should have been going, and ETX should be able to pull off if we can get enough people on board... which is likely to happen given how OTX is slowing down. 

Whilst I may not agree that quad pilots don't care about trims... that in itself shows that customisation is essential... I never used the heli screen (until I did)... but I only need it for that one model. It would be nice to be able to turn tabs off that you don't use... de-clutter the model / screen.

It would be also nice to be able to re-order tabs and even main menu icon order... i.e. I would like the main menu to have the reset option first, so I can reset telemetry/timers between batteries... with basically three presses of the enter key without thinking. But, I'm sure someone would then say - but you could also do what with a special function, or a lua script... meaning there are several different ways to acheive the same goal, depending on your skill level and how you want it to function. 

Just as I'd like to be able define what happens with a long press of buttons like MDL (short press, model setup, long press, channel monitor). Short press of TELE, dedicated panel - set it up as you wish. Long press, panel setup. 

In other words, being able to customise the radio to the nth degree! ;)

raphaelcoeffic, 2021-06-07:
> It would be also nice to be able to re-order tabs and even main menu icon order... i.e. I would like the main menu to have the reset option first, so I can reset telemetry/timers between batteries...

It would probably make sense to have this (and maybe more) on the "TELEM" button directly as a shortcut. What do you think?

pfeerick, 2021-06-07:
Yeah, I could see that as being part of the more appropriate use of the TELEM button... meaning we could move the UI/widgets settings back to being a long press. Would have to work out what else is shown there. I (perhaps due to legacy), prefer a dedicated panel for telemetry, accessible by button... so my suggestion would be that a short press of telem would 'simply' take you to a dedicated panel... meaning it could have multiple widgets, or run a pre-determined full screen lua app.

raphaelcoeffic, 2021-06-07:
> I (perhaps due to legacy), prefer a dedicated panel for telemetry, accessible by button... so my suggestion would be that a short press of telem would 'simply' take you to a dedicated panel... 

that dedicated panel could a good overview of the current values, and offer to reset the telemetry data and/or other options as we see fit.

resetting the timers is however not really “telemetry”. Not sure yet how to handle that properly.

lshems, 2021-06-07:
Please minimize the standard use of shortcuts and shorthand.

The reset telemetry popup is really annoying for people who don't use it
and interferes with the button control that should be hand 'ed over in a
script run scenario.

But to have them available to set as an option, yes.

raphaelcoeffic, 2021-06-07:
> The reset telemetry popup is really annoying for people who don't use it
and interferes with the button control that should be hand 'ed over in a
script run scenario.

Understood. However, the "LUA-keys" issue will be handled differently:
- In the normal "main screen", things stay as they are (no keys passed).
- In the "Tools/App" mode, you will get all the keys / touch and these "key shortcuts" (MDL / SYS / TELEM) are not available, so LUA scripts can do whatever they please with these keys.

The current plan discussed right now is to remove that pop-up and just have a real "Telemetry" screen started by the "TELEM" key (short press), while the screen setup goes back to "long [TELEM] press".

lshems, 2021-06-07:
Well, then I have the same requirement as the drone guys. Where they don't want three presses to reset, I don't want three presses and scrolling to get to my standard telemetry screen I always have on top. I don't even use the opentx GUI apart from changing models and  model settings.

raphaelcoeffic, 2021-06-07:
> Well, then I have the same requirement as the drone guys. Where they don't want three presses to reset, I don't want three presses and scrolling to get to my standard telemetry screen I always have on top. I don't even use the opentx GUI apart from changing models and model settings.

This is perfectly legitimate and is addressed by @JimB40 ‘s design. But I’m sorry it will have to wait a little bit longer, basically until 2.4.1, which is due end of September.

JimB40, 2021-06-07:
Dont't be sorry @raphaelcoeffic. This agile approach in code dev & kazein approach in project management will pay-off. I can name numerous big companies that benefit from it (with Fujifilm on the top as I'm photographer). As we discussed on Sunday it's far better to release smaller improvements every quarter than promise revolution in 2 years. :)

raphaelcoeffic, 2021-06-07:
> than promise revolution in 2 years. :)

Especially if that take 4 at the end. Just saying :-D

lshems, 2021-06-07:
Don't apologize. That's out of the order.

Why don't we categorise the requests according to MOSCOW: must, should, could and wished for requirements.

This would be a should, or perhaps even could. But certainly not a must.

By using such a classification, you can manage expectations.

That helps a lot perhaps.

pfeerick, 2021-06-08:
It's a bit hard here since this is a catch all UI/UX topic. More like, if we can come up with a concrete enough proposals, they can be proposed in their own issue, discussed further, and then be tagged for a future milestone if it is deemed worthy ;) I'm basically not going to raise anything *new* though until say a week past 2.4.0 release, in order to focus on ironing out any remaining kinks in the current codebase, and then start thinking about 2.4.1 ;)

lshems, 2021-06-10:
So what do we do? I posted a proposal on the discussion tab. Should we copy this discussion over there and close the issue? As it is not an issue??😊

https://github.com/EdgeTX/edgetx/discussions/174

raphaelcoeffic, 2021-06-10:
> So what do we do? I posted a proposal on the discussion tab. Should we copy this discussion over there and close the issue? As it is not an issue??😊
> 
> #174

I’ll leave it up to @JimB40 to handle this.

pfeerick, 2021-06-11:
I can see if 'Convert to discussion' does what I think it does if @JimB40 has no objections... then we can keep the _discussion_ going there... what... nobody laughed at that bad pun? :laughing:

JimB40, 2021-06-11:
Would be great Peter. As psychology is a must in people photography I can see one more advantage. 
'Issue' is something connected with misbehaving or problem. Going through 'issues' we unconsciously prepare our minds for problem solving. New ideas are not 'problems' :) These type of stuff should be in different area if github allows.

closed 2021-06-11 by pfeerick