EdgeTX/edgetx #154: EdgeTX splash: 212x64 and 128x64 needed

raphaelcoeffic, 2021-06-06:
Some testers have complained that we don't have any splash screens for the following screen types:
- 4-bit greyscale `212x64` (X9D).
- 1-bit black/white `128x64` (X7, T-Lite and pals).

Please submit your designs here!

MRC3742, 2021-06-07:
Here is just a simple change from the OpenTX logos.

[image]

[image]

Or possibly inverted for a similar but unique look!

[image]

[image]

pfeerick, 2021-06-07:
Since these are monochrome screens ... The inverted look is probably better.. but either/or look good! :)

pfeerick, 2021-06-07:
In addition to the splash screen, there is one more place that logo is needed... on the X9(D) series at least, the default model image is the FrSKY logo with OPENTX underneath ;) Aspect ratio looks like it's just a smaller version of the splash screen.

raphaelcoeffic, 2021-06-07:
@MRC3742 @pfeerick I just created a PR to try these out. I selected the inverted ones for now (white background).

pfeerick, 2021-06-07:
Jesus... you have good timing... I've just reset my X9D+ back to OTX, and was about to let ETX loose on it again... now I'll wait for the PR to finish building! ;)

MRC3742, 2021-06-07:
I could update the logo.png as well for the 212x64 screen later today.

Glad to hear that these splash screens are what you were looking for.
Without access to the font for "EDGE" I had to draw it from the 480x272 image.
I could try to clean it up a little better if it looks choppy on the 212 screen.
The 128 screen is going to be a little choppy with only two colors to work with.

Rich

raphaelcoeffic, 2021-06-07:
> I could update the logo.png as well for the 212x64 screen later today.

Thx, I just did the resizing, let's see how that works.

PR #161 (2021-06-09): Test new EdgeTX splashes and remove vendor specific branding
Resolves #154
commit afb297b0 (2021-06-09): Test new EdgeTX splashes and remove vendor specific branding

Resolves #154
closed 2021-06-09 by pfeerick