EdgeTX/edgetx #158: Add "change sign" button to NumberKeyboard constructor

eshifri, 2021-06-06:
This is especially important for the radios without buttons (e.g. NV14).
It takes a lot of clicks to to change the sign of the value (e.g. from -80 to +80).
The NumberKeyboard screen has enough space to add  additional virtual button "=/-".

Something like this:

NumberKeyboard::NumberKeyboard():
  Keyboard<BaseNumberEdit>(KEYBOARD_HEIGHT)
{
  new TextButton(this, { LCD_W/2 - 115, 10, 50, 30 }, "<<",
                 [=]() -> uint8_t {
                   if (field) {
                     field->setValue(field->getValue() - 10 * field->getStep());
                   }
                   else putEvent(EVT_VK(VKEY_DEC_LARGE));
                   return 0;
                 }, BUTTON_BACKGROUND | BUTTON_NOFOCUS);

  new TextButton(this, { LCD_W/2 - 55, 10, 50, 30 }, "-",
                 [=]() -> uint8_t {
                   if (field) {
                     field->setValue(field->getValue() - field->getStep());
                   }
                   else putEvent(EVT_VK(VKEY_DEC));
                   return 0;
                 }, BUTTON_BACKGROUND | BUTTON_NOFOCUS);

  new TextButton(this, { LCD_W/2 + 5, 10, 50, 30 }, "+",
                 [=]() -> uint8_t {
                   if (field) {
                     field->setValue(field->getNextValue());
                   }
                   else putEvent(EVT_VK(VKEY_INC));
                   return 0;
                 }, BUTTON_BACKGROUND | BUTTON_NOFOCUS);

  new TextButton(this, { LCD_W/2 + 65, 10, 50, 30 }, ">>",
                 [=]() -> uint8_t {
                   if (field) {
                     field->setValue(field->getValue() + 10 * field->getStep());
                   }
                   else putEvent(EVT_VK(VKEY_INC_LARGE));
                   return 0;
                 }, BUTTON_BACKGROUND | BUTTON_NOFOCUS);

  new TextButton(this, { LCD_W/2 - 115, 50, 50, 30 }, "MIN",
                 [=]() -> uint8_t {
                   if (field) {
                     field->setValue(field->getMin());
                   }
                   else putEvent(EVT_VK(VKEY_MIN));
                   return 0;
                 }, BUTTON_BACKGROUND | BUTTON_NOFOCUS);

  new TextButton(this, { LCD_W/2 + 65, 50, 50, 30 }, "MAX",
                 [=]() -> uint8_t {
                   if (field) {
                     field->setValue(field->getMax());
                   }
                   else putEvent(EVT_VK(VKEY_MAX));
                   return 0;
                 }, BUTTON_BACKGROUND | BUTTON_NOFOCUS);

  new TextButton(this, { LCD_W/2 - 55, 50, 50, 30 }, "DEF",
                 [=]() -> uint8_t {
                   if (field) {
                     field->setValue(field->getDefault());
                   }
                   else putEvent(EVT_VK(VKEY_DEFAULT));
                   return 0;
                 }, BUTTON_BACKGROUND | BUTTON_NOFOCUS);
				 
  new TextButton(this, { LCD_W/2 + 5, 50, 50, 30 }, "+/-",
                 [=]() -> uint8_t {
                   if (field) {
                     field->setValue(-field->getValue());
                   }
                   else putEvent(EVT_VK(VKEY_SIGN));
                   return 0;
                 }, BUTTON_BACKGROUND | BUTTON_NOFOCUS);
}

I can create a pull request if agreed.

raphaelcoeffic, 2021-06-06:
That sounds to me like a reasonable feature. Did you reactivate the virtual keyboard, or is this on the NV14?

eshifri, 2021-06-06:
I did it on NV14.

raphaelcoeffic, 2021-06-06:
> I did it on NV14.

Not bad! I haven‘t tried it yet :-(

eshifri, 2021-06-06:
Not that good either :-). I mean on original NV14 branch: https://github.com/eshifri/Flysky-OpenTX-Test-Branch

raphaelcoeffic, 2021-06-07:
@eshifri it would be good to have you on board to help us drive the integration effort.

eshifri, 2021-06-07:
I will be. What about  @qba667? He is the main expert.

raphaelcoeffic, 2021-06-08:
> I will be. What about @qba667? He is the main expert.

I did contact him, and he replied positively. But he seems to be busy since.

closed 2021-08-01 by pfeerick