EdgeTX/edgetx #159: List of protocols from Multi isn't sorted (anymore)

wimalopaan, 2021-06-07:
Selecting Multi on a TX16s and choosing a protocol, the list of protocols is not alphabeticaly sorted, what makes it hardly useable. Tested with multi-1.3.2.61

raphaelcoeffic, 2021-06-07:
@wimalopaan not sure I can finish this before 2.4.0, but that is at least my intention.

pfeerick, 2021-06-13:
On the X9D+ the protocol list is in index number order (not alphabetical), on the T-Lite it is in alphabetical order, and on the TX16S it's in index number order again rather than alphabetical.

In other words, when reading the `/SCRIPTS/TOOLS/MultiChan.txt` file...
[image]

... I can tell straight away that on the X9D+ and TX16S that Flysky, FrSkyD, DSM and Devo (plus some not shown in that screenshot) will be shown at the top of the list...

kevinkoenig, 2021-07-30:
this problem can be fixed in the UI code by simply creating a crosswalk table and sorting the list contained in STR_MULTI_PROTOCOLS.  I am completely new to edgetx development, but I put together a prototype of this approach and have it working and tested on TX16s.  I hesitate to offer it up, as it was a learning project, and it can be better fixed by fixing the underlying list.  

I would take that up, however, I cant figure out how the /SCRIPTS/TOOLS/MultiChan.txt file is compiled into STR_MULTI_PROTOCOLS. If modifying MultiChan.txt is the right approach, and someone can point me in the right direction, I will chip in.

pfeerick, 2021-07-31:
No, modifying that probably isn't right approach, as it's not part of ETX, but actually part of the MPM Lua package, and should be be updated every time you/the end user updates the MPM firmware, along with the LUA scripts they supply. 

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Lua_scripts/MultiChan.txt

If you can spot why the T-Lite (128x64) *is* alphabetical, whereas the X9D+(212x64) and TX16(colorlcd) aren't, that may help.

rotorman, 2021-07-31:
> On the X9D+ the protocol list is in index number order (not alphabetical), on the T-Lite it is in alphabetical order, and on the
> TX16S it's in index number order again rather than alphabetical.

I do not have T-Lite hw, but I tested this using a simu with current main branch head. Under simulated T-Lite I do not see the protocols in alphabetical order (I chose Internal RF Multi for testing).
[image]
@pfeerick Can you please confim that with T-Lite you really have the protocol list in alphabetical order when you scroll through them?

pfeerick, 2021-08-01:
Last weeks Nightly still had the list in order on the T-Lite, so SIMU is *not* accurate in this respect.

https://youtu.be/tGAhJd4SvJE

rotorman, 2021-08-01:
@pfeerick Thank you for double checking this! Do you have an idea where it could stem from that there is a difference with simu and real hw?

pfeerick, 2021-08-01:
Not yet - I really haven't looked into this yet, and I'm being a bit lazy at the moment, as I probably should backup the radio and try a full wipe, just in case it's something in the EEPROM or MPM itself - as then that's something the simu can't/won't repo properly. 

From what Raphael said when we were working through the issues surrounding showing the options for the various protocols, I think the MPM can to report back the list of protocols it supports, so if that is reported in a different order on the T-Lite (possible, since I don't think I ever updated it's MPM firmware - and don't seem to be able to 'easily' interrogate the current version as unlike the TX16s the current version/status isn't shown - will have to see if there is something in the lua scripts that does show it) that would explain the difference. There are bunch of things around the MPM you can't test well in `simu` since it is it's own thing.

raphaelcoeffic, 2021-08-01:
> From what Raphael said when we were working through the issues surrounding showing the options for the various protocols, I think the MPM can to report back the list of protocols it supports, so if [...]

It does report the supported protocols, but only one-by-one. So you need to set a first one and check the next, until you hit the end of the list, so you can save it. This process could take some serious time, so that we'd need to cache it somehow, and show a progress bar while loading the list.

The intention behind the text file was that you would fetch the protocols using the technique described before to fetch a list that the UI would just read when looking for protos. If that list is absent, it will get back to the "standard way", which is what we have right now.

pfeerick, 2021-08-01:
Here's the "but" though... is my T-Lite a one off, or is the T-Lite in a different order when it's on the actual hardware? It's not anything to do with the `MultiChan.txt` file on the SD card at all, as I just tried again with the SD card ejected, and it was still showing all the protocols, and in alphabetical order. But the X9D+, when treated the same way, shows in the order they would have been listed in the `MultiChan.txt` file. 

Talking of progress bars, I think we may need to seriously consider one for the model select screen... as if you have a bunch of models and images on each one it can take long enough that you think the transmitter has frozen for anything to happen... and I don't even have images on more than 1/3 of the models on it atm. :-O

raphaelcoeffic, 2021-08-01:
> Here's the "but" though... is my T-Lite a one off, or is the T-Lite in a different order when it's on the actual hardware? It's not anything to do with the `MultiChan.txt` file on the SD card at [...]

The T-lite does not have to ask for the complete list in advance, as there is no drop-down. It shows you the protocols one at a time ;-)

raphaelcoeffic, 2021-08-01:
> Talking of progress bars, I think we may need to seriously consider one for the model select screen... as if you have a bunch of models and images on each one it can take long enough that you think [...]

Yes, the first caching can be really long. Also, we should write the thumbnails to disk to avoid to have to re-generate them on the fly all the time. That should make it way faster.

kevinkoenig, 2021-08-04:
I am not sure if it is the right approach but i did an implementation of a class that reads the protocol list, sorts it, and maps it back and forth.  So the protocol number doesn't change from what it was.  It localizes the change to model_setup.cpp.  This is obviously only for color lcds.  I am not sure if this is the right way to solve this PR, but I thought I would post the class, and the modified code for review.  

I am not sure what is the best way to post the code so I will paste it here for now.  The code allocates a bunch of strings once which is not the best for memory usage.

class RFProtocols
{
private:
    const char * protocols;
    int maxProtocols;
    std::vector<std::string> labels;
    uint8_t getProtoStringLength();

public:
    RFProtocols(const char * rfProtocols, int max);
    int findProtocolIndex(int rfProtcolIndex);
    int lookupProtocolIndex(int selectedIndex);
    std::vector<std::string> &getLabels();
};

uint8_t RFProtocols::getProtoStringLength()
{
    return protocols[0];
}

RFProtocols::RFProtocols(const char *protocols, int max)
{
    this->protocols = protocols;
    maxProtocols = max;

    uint8_t len = protocols[0];
    const char * value = &protocols[1];

    for (int i = 0; i < max; i++)
    {
        std::string label = std::string(value, min<uint8_t>(len, strlen(value)));
        labels.emplace_back(label);

        value += len;
    }
    std::sort(labels.begin(), labels.end());
}

std::vector<std::string> &RFProtocols::getLabels()
{
    return labels;
}

int RFProtocols::lookupProtocolIndex(int selectedIndex)
{
    uint8_t len = getProtoStringLength();
    const char *p = &protocols[1];

    const char *protocolLabel = labels.at(selectedIndex).c_str();

    int protocolIndex = 0;
    for (; protocolIndex < maxProtocols; protocolIndex++)
    {
        if (strncmp(p, protocolLabel, min((int)strlen(protocolLabel), (int)len)) == 0)
            break;

        p += len;
    }
    assert(protocolIndex < maxProtocols);

    return protocolIndex;
}

int RFProtocols::findProtocolIndex(int rfProtocolIndex)
{
    uint8_t len = getProtoStringLength();
    const char *protocolLabel = &protocols[1] + len * rfProtocolIndex;

    auto result = std::find_if(labels.begin(), labels.end(), [protocolLabel, len](std::string &a) {
        auto theSame = strncmp(a.c_str(), protocolLabel, max((int)a.length(), (int)len)) == 0;
        return theSame;
    });
    assert(result != labels.end());
    return result - labels.begin();
}

// global pointer intialized once on the first call to update()
RFProtocols *rfProtocols = nullptr;

Then in ModuleWindow update() function in the #if defined(MULTIMODULE)

        // initialize rfProtocols if it is not initialized already.  This is a hack
        // because i dont know how to construct it at the exact right time after the module
        // is loaded
        if (rfProtocols == nullptr)
            rfProtocols = new RFProtocols(STR_MULTI_PROTOCOLS, MODULE_SUBTYPE_MULTI_LAST);

        multiRfProto = rfProtocols->findProtocolIndex(multiRfProto);  // map the selected protocol

       rfChoice = new Choice(
            this,
            grid.getFieldSlot(
                g_model.moduleData[moduleIdx].multi.customProto ? 3 : 2, 0),
            rfProtocols->getLabels(), MODULE_SUBTYPE_MULTI_FIRST,
            MODULE_SUBTYPE_MULTI_LAST, GET_DEFAULT(multiRfProto),
            [=](int32_t newValue) {

              newValue = rfProtocols->lookupProtocolIndex(newValue);  // map it back.

              g_model.moduleData[moduleIdx].setMultiProtocol(newValue);
              g_model.moduleData[moduleIdx].subType = 0;
              if (mmSubProtocol != nullptr) mmSubProtocol->invalidate(); // always null ???
              resetMultiProtocolsOptions(moduleIdx);
              SET_DIRTY();
              update();
              rfChoice->setFocus(SET_FOCUS_DEFAULT);// really valid ???
            });

raphaelcoeffic, 2021-08-05:
@kevinkoenig this looks like one part of the solution. Or at least a good temporary solution. But we should implement fetching that list from the module as well (incl. progress bar while fetching).

kevinkoenig, 2021-08-05:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Okay.  Should I check this in?

raphaelcoeffic, 2021-08-05:
> Okay. Should I check this in? Get Outlook for iOS<https://aka.ms/o0ukef>
> […](#)

Sure go ahead!

kevinkoenig, 2021-08-05:
> > Okay. Should I check this in? Get Outlook for iOShttps://aka.ms/o0ukef
> > […](#)
> 
> Sure go ahead!

great, my first Issue solved!  Not all I need to do is figure out the correct procedure.  I guess I need to create a Pull request?  Is there a writeup of how to do it correctly.?

rotorman, 2021-08-05:
@kevinkoenig 
In short, the steps would be:
1) Setup a build system https://github.com/EdgeTX/edgetx/wiki/Building-and-debugging-EdgeTX
2) fork EdgeTX repo to your own GH account.
3) (very important) you make a new branch for the changes
4) recursively git clone your branch to your local system
5) You add your changes to your branch and test that it works locally
6) You commit your changes (can also be in-between)
7) You push your changes
8) You create a pull request from your branch to main EdgeTX repository

zyren, 2021-08-31:
I found the historical record of MPM. From this point of view, the sorting is performed on MPM and sent to Radio to display the list of valid protocols, because I don’t know the code very well, but I have a long time experience in using it, starting from er9x And opentx1.xx, until now edgetx, and mpm also has a long time experience, so I think that the protocol list displayed in the radio comes from the real-time list sent by the MPM Module, not every update. Maintain the corresponding protocol list in the EdgeTX source code.

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases?after=v1.3.0.95

[image]

pfeerick, 2021-08-31:
Not necessarily, but at some point we need to implement interrogation of the MPM for the supported protocols. This, however, is not the focus of this PR. The focus of this PR is to make the best of what we have now, before work on re-writing the entire MPM integration is done - to bring in the things that were were not simultaneously implemented in both the 2.3 and 2.4 OTX branches - as well as any other improvements can be done through a fresh set of eyes.

closed 2021-08-31 by pfeerick