EdgeTX/edgetx #160: Copyright notice for triangle draw func omitted

olliw42, 2021-06-07:
This is an issue with the drawFilledTriangle() function in libopenui, but libopenui repo doesn't have issues, hence raised here.

The drawFilledTriangle() function originates from code from the AdaFruit gfx library, and a respective notice should have been put in.

That's largely my fault since while I do have such a comment in my original OpenTx-based branch, I was not copying it over to my EdgeTx/libopenui PR, from which it was taken over for another PR which was then merged.

As a guide, in my original branch I have a comment 

```
//THANKS to Adafruit and its GFX library !
</span>https://learn.adafruit.com/adafruit-gfx-graphics-library
```

I think a similar notice should be added.

raphaelcoeffic, 2021-06-07:
I believe this is actually the copyright notice that should be there: https://github.com/adafruit/Adafruit-GFX-Library/blob/master/license.txt

raphaelcoeffic, 2021-06-07:
And as it is BSD, we need to place it is as well somewhere in the “documentation”, which covers the binary re-distribution.

raphaelcoeffic, 2021-06-07:
Resolved here: https://github.com/EdgeTX/libopenui/commit/f2330a3059e84c5ea7cb2ed5991a0a205f3045f9

closed 2021-06-07 by raphaelcoeffic