EdgeTX/edgetx #165: Invers not working on all LCD.drawSomething() functions

lshems, 2021-06-08:
Invers doesn't seem to work on LCD.drawNumber().

Drawtext() seems to work. Timer not checked yet,nor switches or sources.

raphaelcoeffic, 2021-06-08:
@lshems that is very well possible. Could you please get me a list of these functions? The point is, these 2 have been more or less obsoleted in 2.4 at the OS level, so we need to simulate it for LUA.

lshems, 2021-06-08:
Ok, I will. Any reason the text is working?

Also, in opentx I had to write inverse spaces in some cases to get it working for the color screens. It's not obvious how to handle that I think.

But this means it needs to be reworked completely? May be an opportunity to do it right than. How are colours managed now. Still Palette?

raphaelcoeffic, 2021-06-08:
> But this means it needs to be reworked completely? May be an opportunity to do it right than. How are colours managed now. Still Palette?

this is what is planed. The Palette is gone off the gfx lib, so it would be totally possible to have V2 API that offers direct colors.

lshems, 2021-06-08:
That sounds good. people didn't understand the palette concept and the palette items where badly chosen.

I just set the color to what I want, draw, and set it back.

Lol.

I'll make the list.

lshems, 2021-06-09:
Can someone comment and put their view forward:

I think we should ditch the INVERS label, as it does not make sense on color screens.

Instead, it would make sense to add a BACKGROUND option where you can pass the colour of the background for your text. This is much more like for instance windows, where you cannot invert characters either, but you can indicate a background colour.

The BLINK however would be interesting, as that is nice on GUI's to pull attention. 

I also want to try and keep some alignement between color and BW lcd commands, as otherwise script development becomes a disaster. Can we take a decision on the need to be backwards compatible, and in what way?

I would love to just see correct general implementations being developed, supporting BW and colour screens, for all lcd commands.

Backward compatibility could be limited to not eliminating the commands, and as soon as they don't work anymore, just make them void??

raphaelcoeffic, 2021-06-11:
@lshems does #192 provide a suitable fix for you? Could you please test?

lshems, 2023-02-27:
NOP, not working yet. No BLINK nor INVERS.

lshems, 2023-02-27:
Here also: what do you do: functional approach or technical approach??

Technical approach: implement a BLINK and an INVERS function for colour displays.
Functional approach would be to have BLINK and INVERS simply mapped to the colours for in focus text and edited text on the colour radios. This would work with one caveat: you can assign both BLINK and INVERS, but not two colours, so some scripts might give strange results. 

I don't know which colours are now default text, in focus and edit colours??