EdgeTX/edgetx #166: X9D+ EEPROM upgrade still wonky

pfeerick, 2021-06-09:
FrSKY X9D+, x9dp-70379b5.bin (most recent build of PR#161)

Right, so I've been able to reproduce that issue with the X9D+ when migrating from OTX 2.3 => ETX 2.4. 

I fully restored the radio to OTX 2.3...  DFU restore to OTX 2.3 bootloader and fw, recalibrate radio, read radio into companion, copy models from OTX file to new radio config, write it, verify models were fine.

I then upgraded the bootloader on transmitter from x9dp-70379b5.bin (most recent build of PR#161), bootloader upgraded the firmware. Fired it up, let it do the EPPROM conversion. First two models and fourth appear to be perfect. Others load fine, like the third, stick are responsive for 1-2 seconds, and then it seems that all Analogs go non-responsive... everything is zero in the `Radio Setup -> Hardware -> Analogs` and values don't change for any stick/pot/slider input, but the switches are updating. Seventh model (Cessna) has a wonky timer  `ter/` instead of `!SA`

For completeness, I'm going to attach the firmware file used from the action, the OTX file with the models, and both the pre and post EEPROMS. The top three models are my own, the rest were from another flyer who used to use this radio, so don't judge! :-P

[files.zip](https://github.com/EdgeTX/edgetx/files/6620390/files.zip)

raphaelcoeffic, 2021-06-09:
Thx for the files, I'll add them to the unit tests if you don't mind!

raphaelcoeffic, 2021-06-09:
@pfeerick no copyright issues here with the models? 😇

pfeerick, 2021-06-09:
Not as far as I'm concerned... do with them as you wish...

raphaelcoeffic, 2021-06-09:
@pfeerick the "wonky timer" is already fixed, but the "unresponsive sticks" is still a mystery. I might need additional info.

For now, what we know (or think to know):
- model specific (make no sense, the analogs are radio specific).
- take 1-2s to "activate".
- I could not reproduce it in the simulator.

pfeerick, 2021-06-09:
The time aspect is a hard one to pin down... it depends how quickly I can get out of the menu, and get/make sure the stick legends are shown. One factor altering the time *could* (I stress the could) be if an audio file plays like the mode a switch is on it *seemed* to be responsive a smidgen longer. I would have to do some more trials to make sure I'm getting the sequence of events right. I'll also bind up a receiver also tomorrow to see if there is any feedback from the sticks, as well as trying that test build

raphaelcoeffic, 2021-06-09:
@pfeerick it was indeed related to the fact that these models are using the internal XJT module! This module does not trigger the mixer via a timer (like all other modules), but by using an external trigger from the internal XJT module. This in turn prevented the ADC values from being refreshed, which, well...., lead to the exact issue you described.

closed 2021-06-09 by raphaelcoeffic

pfeerick, 2021-06-10:
I can confirm it is fixed, and was internal XJT related. More testing before applying latest firmware suggests that it was also D16 specific, as D8 was working.

commit f7d36cd6 (2021-06-09): Place adcRead() back to mixer task to avoid issues with heartbeat triggered modules

Resolves #166
PR #172 (2021-06-10): Fix "wonky timer"
Resolves #166 (first part of it).