EdgeTX/edgetx #167: X9D+, Mix name garbage - possibly related to new model wizard

pfeerick, 2021-06-09:
FrSKY X9D+, x9dp-70379b5.bin (most recent build of PR#161), still running OTX SD pack. 

Created a new model, using the new model wizard (tested plane and multicopter), accepting all defaults. Now go into model setup -> Mixes  and the mix names on the end are garbage characters. However, skipping the wizard will create just the first four channels in default order specified, and they seem to be correctly blanked. 

Going to test with the new SD card pack in case in the hope the new model wizard is a file from that and is already fixed.

raphaelcoeffic, 2021-06-09:
Very good hint! It seems the LUA names are still translated into ZCHARs then, I'll have a look.

pfeerick, 2021-06-09:
Closed by #171

PR #171 (2021-06-09): ZChar nightmare / take 2
Resolves #167
closed 2021-06-09 by pfeerick