EdgeTX/edgetx #169: MDL/SYS keys' bindings

JimB40, 2021-06-09:
I think we may consider implementing this idea in 2.4. So we'll have initial feedback after release.
If OS wide implementation would affect code too much I'd go with MainView as it is in OpenTX now.  

event -> tab
------------
MDL short -> MODEL SETUP
MDL long -> MODELS (Select Model)
SYS short -> TOOLS
SYS long -> RADIO SETUP (since no APPS yet)

pfeerick, 2021-06-17:
TELE short -> Channel Monitor
TELE long -> UI Setup, or Telemetry Panel once/if that's decided up.

JimB40, 2021-07-28:
@raphaelcoeffic Can we increase priority for this feature? :)
This will speed up testing 2.5 features on radio a lot, and we can have initial feedback on Channel monitor key binding.

With TELE exactly as @pfeerick proposed
TELE short -> channel monitor
TELE long -> UI setup

pfeerick, 2021-07-31:
This PR and some of the discussion therein has some bearing on this... as there were changes around short and long button press handling...

pfeerick, 2021-08-05:
Sorry, forgot include the PR number... #58

JimB40, 2021-11-11:
So what is status now @pfeerick?
That's one of core 2.6/2.7 UI revamp assumptions
https://github.com/EdgeTX/edgetx/discussions/186#discussioncomment-1623591

kevinkoenig, 2021-12-05:
> @raphaelcoeffic Can we increase priority for this feature? :) This will speed up testing 2.5 features on radio a lot, and we can have initial feedback on Channel monitor key binding.
> 
> With TELE exactly as @pfeerick proposed TELE short -> channel monitor TELE long -> UI setup

TELE short will interfere with key processing in Theme Editor for the Color Editor.  Right now that key moves between the color bars.  We need to change that if the decision is to change TELE short to bring up Channel Monitor

pfeerick, 2021-12-05:
It's not such a problem as IMO generally the shortcuts should only work at the main screen... i.e. I don't think MDL a short press should take should take you to model setup if you are already in RADIO menu... or in screen settings, etc... You should exit the current screens, and once you're back at the main screen, you can use them. But I can see an argument for short press of TELE being usable in the model setup screen for channel monitor... but don't particularly care for it as a *need*.

robhaswell, 2022-02-06:
> MDL short -> MODEL SETUP
> MDL long -> MODELS (Select Model)

Can I query why these assignments are this way round? I reckon that pilots switch models more often than they set them up.

I think it would be more intuitive if the MDL button went to MODELS and long press went to SETUP. That would follow the common UX pattern of "short press to select, long press to edit".

JimB40, 2022-02-07:
Good point. Still there are some issues to implement it.

JimB40, 2022-03-27:
@pfeerick 

Regarding key bingings you know my point of view. :)

I would bind SYS and MDL as all-around the system shortcuts that means that whatever is displayed presing them will take you to the binded screen. No mater if you are in MODEL Tabs view, Radio Tabs view, Channel Monitor, Screen setup, any Modal window etc. These should be all-time active shortcuts that allows you quickly navigate to quite distant place in UI tree without "going up" to Main view with Exit and "going down" using Enter

As we already discussed an partly agreed:

MDL short -> goes to MODEL configuraration Tabs set (exactly to the 1st one that is MODEL SETUP now)
MDL long -> goes to MODELS screen (switching, adding, deleting etc)
SYS short -> goes to TOOLS tab which in futute release will be separate screen called APPS
SYS long -> goes to RADIO configuration Tabs set (exactly to the 1st one that is now TOOLS but after making separate screen called APPS and moving SD Card browser to be one of applications 1st screen will be RADIO SETUP and this will be consistent with MDL long to enter configuration)

First three bindings are already implemented thanks to https://github.com/EdgeTX/edgetx/pull/1610 but only on main screen view. If you go to any other place in system, SYS & MDL do nothing. Bindings are "empty"
(TBH it needs to be chceked if someone hasn't used it in any other PR - because they were free)

That leaves TELEM, PAGE > and PAGE< to be used for other purposes.

We have three ideas to use it:

Screen Setup - (TELEM short press now)
TELEMETRY tab in Model configuration
Channel monitor
We can choose two to be all around system shortcuts or code behaviour of TELEM depending on curently displayed content.

Relates to: #1731

pfeerick, 2022-03-28:
> SYS long -> goes to RADIO configuration Tabs set (exactly to the 1st one that is now TOOLS but after making separate screen called APPS and moving SD Card browser to be one of applications 1st screen will be RADIO SETUP and this will be consistent with MDL long to enter configuration)

No that won't... that will be the exact OPPOSITE of the current, and not changing, MDL short press to enter model setup. So those two are back to front. 

> We can choose two to be all around system shortcuts or code behavior of TELEM depending on curently displayed content.

We also have to keep in mind that keys like TELE are also in use in other parts of the UI... i.e. when in the theme editor, TELE key is not available for use. As are MDL and SYS and any other shortcuts when in lua scripts. So those are the times I can think of off the top of my head that otherwise global shortcuts won't work. 

IMO, page keys should not be repurposed, they are your navigation keys. Additionally, you can also only consider them as one button with a short and long press option. as most radios only have the one page key. 

So the moment that leaves MDL and SYS unchanged in their behaviour (although technically the SYS long press behaviour could be added since those tabs exist), but made global. 

So per, https://github.com/EdgeTX/edgetx/pull/1713#issuecomment-1079894974, this still leaves the question of what to assign TELE short and long... and whether it be global or not. As I raised the question of do we actually  want to go to the MDL -> TELE tab, or just the table of telemetry sensors (which would be implemented as a new page)?

JimB40, 2022-03-28:
Just wondering if we are still bleedin edge. :)
How to intrododuce something new if there is always argument of 'I get used to the way it was in OpenTX'
Henry Ford said one day asked why he does not ask users about his new product.
"We asked and users want faster horse carriages"

Cross-navigation-tree shortcut always speeds up UX and that's a fact. 
Second fact is that changes to UX are always push user to asdjust to a new way. Something you used to call 'muscle memory'.

1. We can make branch and test
2. We can make UI option to assign bindings of SYS & MDL
3. We can change TELEM assignmet in themes as we set it this way.
4. We can unbind all keys if LUA script is running passing all keys events overe there.

But IMHO we can't say it's good how it was because then progress is stopped. 

Just my 5 cents.

pfeerick, 2022-03-28:
"Bleeding edge" does not mean "start implementation without outlining everything that is required so that the person implementing knows what the heck they are doing". 😛 I'm trying to get a concrete outline from you as to exactly how this looks, so there is "oh, I meant it should look like this" later. 😉 If time is going to be spent on implementing this, then the scope and outline of how it should look and work needs to be clear cut. 

1- ~~I thought we were just going to commit this to main untested!~~ 😆 
2 - Feature creep!!! Yes, I want that also... but that comes afterwards unless you want this to be a 3.0 feature!!!
3 - Yup, so what is the proposal for a an alternative for it? That needs to happen at the same time. 
4 - No, they would never be "bound" in the first place... it is a matter of going though the code in multiple places and binding the keys to that function... 

> changes to UX are always push user to adjust to a new way. Something you used to call 'muscle memory'.

Given this is a only a UX change, I'm not so worried about the loss of muscle memory. Especially when if this is done _properly_, either people will see the benefits, or they will hang out for the next release which allows it to be configurable.

JimB40, 2022-03-28:
So let's do what this feature request is about -> SYS & MDL (short and long) key bindings 
Discussing about TELEM or if I curse pressing Delete instead of Edit will not bring solution.
We can test it on Radio tabs set and Model tabs set. (without modal windows - editing)

PR #4111 (2023-10-16): feat(color): Improved navigation with SYS and MDL keys
Allow SYS and MDL key navigation from within radio and model setup pages.

In the Radio Setup pages:
- MDL navigates to the Model Setup page
- Long press MDL navigates to Manage Models page

In the Model Setup pages:
- SYS navigates to the Radio Setup (TOOLS) page
- Long press SYS navigates to the Radio Setup (SETUP) page
- MDL navigates to the Channel Monitor (existing function)
- Long press MDL navigates to the Manage Models page

In the Channels Monitor page:
- MDL navigates to the Model Setup page
- Long press MDL navigates to the Mana [... cut]
PR #4219 (2023-10-31): feat(color): Improved hardware key navigation.
Fixes #169 

As suggested in #4111, this PR extends the key navigation changes.
- Add TELE long press (open Channel Monitor)
- SYS, MDL and TELE keys can navigate freely between Main View, Radio Setup, Model Setup, Channel Monitor, Manage Models and Screen Setup pages.
closed 2023-10-16 by pfeerick