EdgeTX/edgetx #170: Adding whitespace between parameters

JimB40, 2021-06-09:
This another thing we may consider to add to 2.4

Currently going to any MODEL/RADIO tab we have (depending on TAB) 9 parameters lines, so it's quite tight. 
In terms of touchUI it creates errors due to too small vertical space.

I'd go with 8 lines visible (2-3 px) . Maybe 7 lines but then we may loose some 'data overview' advantage.

raphaelcoeffic, 2021-06-09:
> I'd go with 8 lines visible (2-3 px) . Maybe 7 lines but then we may loose some 'data overview' advantage.

We can try different settings and find the sweet spot.

raphaelcoeffic, 2021-06-09:
@JimB40 I promised someone to give it a shot for 2.4.0, but if I don't succeed fast, I'd rather move it to 2.4.1.

JimB40, 2021-11-11:
As stated in schedule we should try to address it in 2.6
This UI element height is kind general setting?
Or every UI field/element on every TAB needs to accessed separately?

JimB40, 2021-11-11:
Connects & solves #376

raphaelcoeffic, 2021-11-11:
> As stated in schedule we should try to address it in 2.6 This UI element height is kind general setting? Or every UI field/element on every TAB needs to accessed separately?

Somehow general. However, it is linked to the line height and centering inside the fields.... So a bit of tweak is necessary to make it work generally.

gagarinlg, 2023-01-17:
Is this still valid with LVGL and the upcoming 3.0 UI changes?

gagarinlg, 2023-03-18:
fixed with LVGL and also stale

closed 2023-03-18 by gagarinlg