EdgeTX/edgetx #178: !SC and !D are missing on the T-Lite / !SH is also missing from the X9D+ and TX16S

raphaelcoeffic, 2021-06-10:
Mentioned in #175

raphaelcoeffic, 2021-06-10:
@pfeerick I think this is on purpose for 2 positions switches, as it is considered redundant. This is true, but a bit confusing at the same time, as one wonders why the switch are not in that list.

bottom line: we can change this very easily, but it is probably a matter of taste. I leave this to you.

raphaelcoeffic, 2021-06-10:
Here is for reference the relevant code:
```
    frame #0: 0x00000001000e5aa6 simu`isSwitchAvailable(swtch=24, context=MixesContext) at gui_common.cpp:322:11
   319 	      return false;
   320 	    }
   321 	    if (!IS_CONFIG_3POS(swinfo.quot)) {
-> 322 	      if (negative) {
   323 	        return false;
   324 	      }
   325 	      if (swinfo.rem == 1) {
```

pfeerick, 2021-06-11:
Ah... ok... that makes sense... the ! states are important for the 3POS switches, but are redundant... I've give that some thought then, and if there's valid reasons to add them as well, I'll do a PR. I'll leave this issue open, and target it to be resolved one way or the other in 2.4.1 - as I think it either needs to be documented, or allowed.

pfeerick, 2021-08-25:
Verdict of the poll was 

[image]

so it looks like this is going ahead...

PR #643 (2021-08-30): Don't remove 2POS from avail switches for invert
Resolves #178 by removing the blocking of negative states for 2POS switches. Only `simu` tested at present.

Also prevent 'unused variable' compiler warning that results from change. 

Remo
closed 2021-08-30 by pfeerick