EdgeTX/edgetx #179: Consider adding simulator support for JSx and JSy in Horus X12S radios.

Miami-Mike, 2021-06-10:
If you want to use the joystick at the lower-left corner of a Horus X12S, it shows up as input selections "JSx" and JSy", and I'm told they work properly, but the joystick isn't simulated in Companion. 

See [Companion use Jsx and Jsy #5993](https://github.com/opentx/opentx/issues/5993) in the OpenTX forum. The issue was raised three years ago and eventually marked stale, and I'm not aware of any sign that the OpenTX developers plan to do anything more. Kilrah says "There is currently no way to simulate them."

raphaelcoeffic, 2021-06-10:
@Miami-Mike companion will have it’s time, but it‘s not yet arrived ;-) thx for documenting though.

elecpower, 2023-07-24:
@Miami-Mike can you check and advise if this is fixed in 2.9 RC 1