EdgeTX/edgetx #181: Ability to sort models in 'folders' / categories

pfeerick, 2021-06-11:
OTX 2.3 supported the concept of having models in folders/categories... which were shown at the side of the screen:

[image]

OTX/ETX 2.4 has the same concept... except they are shown at the top of the screen:
[image]

In OTX2.3, there was the option to move models, as well as create, rename and delete categories:
[image]

These are all missing from ETX:
[image]

I believe the categories are managed via section headings in `RADIO/models.txt`, so in the interum, until this is exposed in the GUI again, it's possible to manipulate folders via that file - it's just a matter of figuring out which bin belongs to which model (shown on the bottom line of the model select screen)

i.e. 
```
[Multirotor]
model1.bin
model2.bin
model3.bin
model4.bin
model5.bin
model6.bin
model7.bin
model8.bin
[Fixed Wing]
model9.bin
model10.bin
model11.bin
[Helicopter]
model12.bin
model13.bin
[Other]
model14.bin
model15.bin
model16.bin
model17.bin
model18.bin
model19.bin
```

JimB40, 2021-06-11:
FYI we had brief discussion with @raphaelcoeffic abt "models' groups" functionality while addressing new MODEL UI. Very initial conclusion was this may be abandoned it in favor of "sorting & filtering" plus "favorite models" functionality.

lshems, 2021-06-11:
Labels might be nice for filtering. Added in the models.txt.
That we could structure as a table to load also in Lua easily.

 I think his is also linked to the "backup" issue and exchange of models without companion. 

I use a Lua to adapt the models.txt and create a reference to a model file I copy over. But this really should be possible in a GUI supported way.

New issue?

pfeerick, 2021-06-11:
> New issue?

Nah, it's relevant to this also, especially if there is a preference to ditch folders/categories entirely... I'd like to see the models.txt ditched entirely, making the bin file self-contained... copy the bin to another tx, and away you go... either that, or maybe an export/import system would be better, since you probably want to package the images and sounds, and potentially any lua scripts etc, with the model.

sfjuocekr, 2021-07-09:
I was also missing the move and category functionality!

On my radio I have made groups for planes, quads, helis, boats and cars. This can still be done manually as explained above by editing models.txt.

Maybe an even better idea is to use labels instead, make the UI show any labels you have like the categories now do and ditch the more or less static file references like models.txt and the convoluted modelXYZ.bin naming for UUIDs.bin which can just be parsed from the models directory?

closed 2021-09-18 by pfeerick