EdgeTX/edgetx #182: USB SD has a delay before working if USB Joystick used beforehand.

pfeerick, 2021-06-11:
TX16S, gh action build for head, https://github.com/EdgeTX/edgetx/commit/4627b3c3240844dd3e07f06d845565c7cb9debd4

As you will be able see from the log below, I connected in joystick mode, then unplugged the USB, and then connect it again. There was then a hefty delay before it would connect (the transmitter straight away showed the USB storage log), with the two device descriptor read errors (which do not normally show up).

The first read error appears nearly immediately, then the second just before it finally ennumerates. It occurs regardless which mode is selected first, and also if it is a second connection in the same mode. i.e. USB Storage -> disconnect -> USB Stroage -> delay still occurs. 

```
[ 8138.092641] usb 3-4: new full-speed USB device number 18 using xhci_hcd
[ 8138.234179] usb 3-4: New USB device found, idVendor=1209, idProduct=4f54, bcdDevice= 2.00
[ 8138.234191] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8138.234196] usb 3-4: Product: RM TX16S Joystick
[ 8138.234199] usb 3-4: Manufacturer: OpenTX
[ 8138.234202] usb 3-4: SerialNumber: 00000000001B
[ 8138.236286] input: OpenTX RM TX16S Joystick as /devices/pci0000:00/0000:00:1c.0/0000:07:00.0/usb3/3-4/3-4:1.0/0003:1209:4F54.000A/input/input36
[ 8138.236689] hid-generic 0003:1209:4F54.000A: input,hidraw5: USB HID v1.11 Gamepad [OpenTX RM TX16S Joystick] on usb-0000:07:00.0-4/input0
[ 8144.758080] usb 3-4: USB disconnect, device number 18
[ 8149.979401] usb 3-4: new full-speed USB device number 19 using xhci_hcd
[ 8153.896101] usb 3-4: device descriptor read/64, error -71
[ 8169.512838] usb 3-4: device descriptor read/64, error -110
[ 8169.742790] usb 3-4: new full-speed USB device number 20 using xhci_hcd
[ 8169.887455] usb 3-4: New USB device found, idVendor=0483, idProduct=5720, bcdDevice= 2.00
[ 8169.887467] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8169.887472] usb 3-4: Product: RM TX16S Mass Storage
[ 8169.887476] usb 3-4: Manufacturer: OpenTX
[ 8169.887479] usb 3-4: SerialNumber: 00000000001B
[ 8169.889664] usb-storage 3-4:1.0: USB Mass Storage device detected
[ 8169.889818] scsi host8: usb-storage 3-4:1.0
[ 8170.903626] scsi 8:0:0:0: Direct-Access     RM_TX    RM TX16SRadio    1.00 PQ: 0 ANSI: 2
[ 8170.903996] scsi 8:0:0:1: Direct-Access     RM_TX    RM TX16SRadio    1.00 PQ: 0 ANSI: 2
[ 8170.904479] sd 8:0:0:0: Attached scsi generic sg6 type 0
[ 8170.904659] sd 8:0:0:1: Attached scsi generic sg7 type 0
[ 8170.904742] sd 8:0:0:0: [sde] 3862528 512-byte logical blocks: (1.98 GB/1.84 GiB)
[ 8170.905000] sd 8:0:0:1: [sdf] 4108 512-byte logical blocks: (2.10 MB/2.01 MiB)
[ 8170.905162] sd 8:0:0:0: [sde] Write Protect is off
[ 8170.905169] sd 8:0:0:0: [sde] Mode Sense: 00 00 00 00
[ 8170.905312] sd 8:0:0:1: [sdf] Write Protect is off
[ 8170.905315] sd 8:0:0:1: [sdf] Mode Sense: 00 00 00 00
```

raphaelcoeffic, 2021-06-11:
I’m afraid this is something we will have to live with a bit. This is probably linked to the USB drivers from ST we are using. USB debugging is really hard work... I would document it for now in the known issues.

pfeerick, 2021-06-12:
Additionally, there seemed to be less of a delay on Windows... so it could be related to the driver stacks also...

ThomasKuehne, 2024-04-26:
Is that issue TX16S / X9D+ sepcific or maybe already fixed?

Using current e3b56b7feabb7c3bb2c6fbf01ec7a2b9f6ea033e with a TLite reconnects without any delay or error messages:
```
[ 4589.784354] usb 1-4: new full-speed USB device number 15 using xhci_hcd
[ 4589.934422] usb 1-4: New USB device found, idVendor=1209, idProduct=4f54, bcdDevice= 2.00
[ 4589.934438] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4589.934445] usb 1-4: Product: Jumper TLite Joystick
[ 4589.934450] usb 1-4: Manufacturer: OpenTX
[ 4589.934455] usb 1-4: SerialNumber: 00000000001B
[ 4589.937264] input: OpenTX Jumper TLite Joystick as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:1209:4F54.0009/input/input32
[ 4589.937724] hid-generic 0003:1209:4F54.0009: input,hidraw1: USB HID v1.11 Gamepad [OpenTX Jumper TLite Joystick] on usb-0000:00:14.0-4/input0
[ 4629.392194] usb 1-4: USB disconnect, device number 15
[ 4630.532377] usb 1-4: new full-speed USB device number 16 using xhci_hcd
[ 4630.682256] usb 1-4: New USB device found, idVendor=1209, idProduct=4f54, bcdDevice= 2.00
[ 4630.682271] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4630.682279] usb 1-4: Product: Jumper TLite Joystick
[ 4630.682284] usb 1-4: Manufacturer: OpenTX
[ 4630.682289] usb 1-4: SerialNumber: 00000000001B
[ 4630.685346] input: OpenTX Jumper TLite Joystick as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:1209:4F54.000A/input/input33
[ 4630.685884] hid-generic 0003:1209:4F54.000A: input,hidraw1: USB HID v1.11 Gamepad [OpenTX Jumper TLite Joystick] on usb-0000:00:14.0-4/input0
```

The only condition where I could reproduce any delay was if both times SDCard mode was used. That delay however has nothing to do with the radio's USB layer. It is caused by the storage auto mount function on the PC.