EdgeTX/edgetx #185: Ability to disable and enable touch screen (to prevent unwanted input)

pfeerick, 2021-06-11:
The question has been asked once or twice, so formalising it here:

Basically, some way of making it so the touch screen is disabled to prevent unwanted input. It could be through LUA, through an option that requires a double tap to enable, an enable/disable option in the radio setup, or perhaps a button on the UI that once you press it, a double tap or a tap and hold, or something like that, enables touch again. Just spit-balling some options.

https://discord.com/channels/839849772864503828/839895574244229152/852756283978416139
> oryfoxer — Today at 1:49 PM
> im kinda concered I can do something stupid mid flight when I have my goggles on
> Is there a way to have the screen require a double touch or something to activate?
> when i finish my flihgt, im frequently editing a widget

NemesisXB, 2021-06-11:
+1 for this.

Maybe long press the return button to disable and long press again to enable.
Or better, the ability to assign it to a button via a global function like we do with the volume.

nangel0, 2021-06-13:
Why not a timer as for backlight? With same behavior like when we have backligh set to come on with KEYS?

gagarinlg, 2021-11-06:
Is here anything open, or can it be closed?

closed 2021-11-06 by raphaelcoeffic