EdgeTX/edgetx #196: Can't select MPM subtype

pfeerick, 2021-06-13:
With the MPM, some protocol types have a subtype, and on the T-Lite (430902ae) and X9D+ (430902ae) it is not possible to select the subtype. When you scroll down, it beeps (suggesting at the end of the list) and when you press up, nothing happens, with a single or repeated presses. Trying a model that was imported from OTX, it would seem that you can scroll down the list (as it was using a subtype that wasn't the last entry), but not back up. Does not effect TX16S, appears B&W specific. 

To reproduce, select MPM for either the Internal or External RF (it effects both), set the type to something like DSM or FrSky, and try to change the subtype . 

Was [reported on Discord](https://discord.com/channels/839849772864503828/839895574244229152/853251821828308993), and videos showing buggy and expected behaviour here:

ETX Bugged behaviour:
https://www.youtube.com/watch?v=r5CV2sLziHU

OTX Working behaviour:
https://www.youtube.com/watch?v=mrXk8p23iyo

pfeerick, 2021-06-13:
It could just be me also, but I think the up and down buttons scrolling through those lists the wrong away around on the T-Lite... when scrolling through the protocol types... T-Lite up button is going down the list (A -> Z), and down button is going up the list (Z -> A). On the Taranis, it could be argued it is doing the right thing, since the names are not in numerical order but protocol order on it, and + is increasing the index, and - is decreasing.

raphaelcoeffic, 2021-06-13:
>  T-Lite up button is going down the list (A -> Z), and down button is going up the list (Z -> A).

could this be the explanation for #159 ?

pfeerick, 2021-06-13:
No, that's a different problem... re #159 on the X9D+ the protocol list is in index number order (not alphabetical), on the T-Lite it is in alphabetical order, and on the TX16 it's in index number order again rather than alphabetical. 

I'll just repeat that there also ;)

raphaelcoeffic, 2021-06-13:
> No, that's a different problem... re #159 on the X9D+ the protocol list is in index number order (not alphabetical), on the T-Lite it is in alphabetical order, and on the TX16S it's in index number order again rather than alphabetical.

What a mess!

raphaelcoeffic, 2021-06-15:
@pfeerick the alphabetical order is given by the module itself (maybe based on FW version?), whereby it uses numerical order when this is not available. I don't have a X9D+ to test with but this is what Pascal explained me.

raphaelcoeffic, 2021-06-15:
@pfeerick The subType issue should now be fixed in #217. Could you please have a look? It should work equally on T-LITE and X9D+ (I mean, subtype should be avail. on both).

pfeerick, 2021-06-15:
> @pfeerick the alphabetical order is given by the module itself (maybe based on FW version?), whereby it uses numerical order when this is not available. I don't have a X9D+ to test with but this is what Pascal explained me.

That is quite possible... I don't have a MPM module in the X9D+, so that would explain it's behaviour. The T-Lite is obviously internal, leaving the TX16S/colour radios the oddballs still. Downloading now... ;)

raphaelcoeffic, 2021-06-15:
> That is quite possible... I don't have a MPM module in the X9D+, so that would explain it's behaviour. The T-Lite is obviously internal, leaving the TX16S/colour radios the oddballs still. [...]

I think it has just not been implemented on the new UI…0

pfeerick, 2021-06-15:
Why does that not surprise me... In other news... #217 indeed makes scrolling both up and down the list work on the X9D+ and T-Lite... which is good, as I buggered up a model I was going to fly before by changing to protocol subtype, so don't have to restore the backup for it now! :grinning:

PR #217 (2021-06-15): Fixed MPM proto subtype reset on subtype change
Resolves #196
closed 2021-06-15 by raphaelcoeffic