EdgeTX/edgetx #197: Numeric fields edit navigation bug

JimB40, 2021-06-13:
**Describe the bug**
Pressing RTN while in edit-mode closes view instead of going back to field focused state.
As far as I tested only numerical fields are affected and not all of them. If field id in group it works as expected

**To Reproduce**
1. Goto edit mode with any not grouped numerical field
2. Press RTN

**Expected behavior**
Pressing RTN while in edit-mode should go back to field focused state.

**General information**
 - EdgeTX 2.4   d99e7a04
 - RM TX16S

pfeerick, 2021-06-21:
i.e. `Sys -> Radio Setup` and pressing `ENT` on any of the date or time fields to select it. Pressing `RET` jumps out of the menu rather than out of the field.

eshifri, 2021-06-21:
It is the same behavior for sliders.

eshifri, 2021-06-22:
The pull request did not go through.

The change is in the libopenui.

The fix is to add

  case EVT_KEY_FIRST(KEY_EXIT):
    return;

after line 104 in numberedit.cpp

commit 063eef1e (2021-06-23): Updated libopenui

Resolves #197
closed 2021-06-23 by raphaelcoeffic