EdgeTX/edgetx #203: SBUS Trainer in reboots radio

rotorman, 2021-06-13:
Tested with RM TX16S 9faea55a with internalgps option built in.

HeadTracker with SBUS output ( https://github.com/dlktdr/HeadTracker/wiki/SBUS-Output ) connected to TX16S AUX1.
SYSTEM -> HARDWARE -> Serial port set to SBUS Trainer
Going in MODEL -> MODEL SETUP -> Trainer Mode and selecting Master/Serial, immediately reboots the radio.

rotorman, 2021-06-14:
Retested with 26b15d4d, where INTERNAL_GPS and UNEXPECTED_SHUTDOWN were not specified and nothing connected to AUX1.
Same result, when SYSTEM -> HARDWARE -> Serial port is set to SBUS Trainer, as soon as under MODEL -> MODEL SETUP -> Trainer Mode is selected to Master/Serial -> Emergency Mode is shown.

rotorman, 2021-06-14:
Hooked up to Segger Ozone via SWD, interestingly no HardFault or error on reboot, debugger does not even abort, the execution just continues after restart (with Ozone, have the UNEXPECTED_SHUTDOWN=NO built in).
Need more investigation what happens here.

raphaelcoeffic, 2021-06-14:
> Hooked up to Segger Ozone via SWD, interestingly no HardFault or error on reboot, debugger does not even abort, the execution just continues after restart (with Ozone, have the UNEXPECTED_SHUTDOWN=NO built in).
> Need more investigation what happens here.

This is probably a watchdog reboot. Does it take 1-2 sec until it reboots? If yes, then watchdog.

rotorman, 2021-06-14:
Nope, unfortunately the reboot is pretty immediate, when ran without stepping.
Due to lack of trace I have not yet pinpointed it.

Until so far I see that, after selecting Master/Serial, in aux_serial_driver.cpp auxSerialInit() is called and UART_MODE_SBUS_TRAINER() case is executed. It then lands in auxSerialSetup() where it reboots. It looks to me as if auxSerialSetup() is executed twice and on the second run it reboots.

rotorman, 2021-06-14:
No additional hw is required to test this: set SYSTEM -> HARDWARE -> Serial port to SBUS Trainer, and as soon as under MODEL -> MODEL SETUP -> Trainer Mode is selected to Master/Serial, at least with RM TX16S you should have a reboot or Emergency Mode (according to UNEXPECTED_SHUTDOWN cmake option).

pfeerick, 2021-06-14:
TX16S, Emergency Mode, recent https://github.com/EdgeTX/edgetx/pull/193 build. 
X9D+, Master/Serial is not an Model setup option
T-Lite, Doesn't have a serial option

raphaelcoeffic, 2021-06-14:
This seems to be fixed for me with f3fbed5a1bad. Not sure why exactly, but we'll see.

rotorman, 2021-06-14:
Solved by f3fbed5. Thx @raphaelcoeffic !

commit 6b596602 (2021-06-16): Fixed trainer special function parameter assignement

Resolves #203
closed 2021-06-14 by rotorman