EdgeTX/edgetx #204: Horus X12S Internal GPS

andytransonic, 2021-06-14:
*Before suggesting a new feature here, please search the open issue if there is already a request for the feature*

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.
 X12S Internal GPS support
 Internal hour auto-adjust from GPS date / time
 For auto-adjust, timezone setting should be in increments of quarter hour (For Nepal it's 5Hrs:45 mins)

Luznatural, 2021-11-07:
Not only for autoadjust from GPS date/time, but for many other features: Distance to the plane, angle-off, plane localizing, etc

raphaelcoeffic, 2021-11-07:
> Not only for autoadjust from GPS date/time, but for many other features: Distance to the plane, angle-off, plane localizing, etc

AFAIK, internal GPS is already supported on X12S. This issue is really just about being able to set the timezone properly.

raphaelcoeffic, 2023-09-14:
Implemented in #3406

closed 2023-09-14 by raphaelcoeffic