EdgeTX/edgetx #206: Auto power off no longer working if warning trigger is cleared

pfeerick, 2021-06-14:
On the TX16S, if you attempted to power down and the receiver was 'still connected', you would get the shutdown confirmation prompt. Disconnecting the power to the receiver would then cause the radio to auto-power off. This is no longer happening.

Similar, with USB HID (Joystick) and USB Storage (SD), disconnecting the USB no longer triggers the radio to continue powering off. 

Does not effect T-Lite - it behaves as expected. 

Behaviour was exhibited when testing latest builds for https://github.com/EdgeTX/edgetx/pull/200 and https://github.com/EdgeTX/edgetx/pull/193

raphaelcoeffic, 2021-06-16:
I think we have an issue here: if you confirm shutdown with USB plugged, it does not really shutdown, but is caught in some bootloop that we cannot really catch (Power is maintained because of USB....), so we should better prevent you from shutting down with USB plugged (without possible confirmation).

pfeerick, 2021-06-16:
Ah, yes, you have a good point there... as I noticed when you did pull the USB after forcing the power off, the LCD finally went dim... i.e. it didn't fully power off. This is still different to the actual issue at hand though, which is that the ... auto continue (for want of better word) is not happening if the reason for the warning is removed... ie. receiver is powered off, USB is disconnected.

raphaelcoeffic, 2021-06-16:
@pfeerick that is true, but I need a bit more time to make it correct ;-) So let's call it a "known issue" for now, and make it correctly the next days.

nangel0, 2021-11-07:
Still happening on tx16s with latest master, but works as expected on jumper t12..

PR #1164 (2021-11-28): fix auto power off on cleared warning trigger
Fixes #206 

also adds a close condition callback to the confirmation dialog
closed 2021-11-28 by pfeerick