EdgeTX/edgetx #208: Scroilling eratic when setting limits in custom failsafe mode

phileville, 2021-06-14:
**Describe the bug**
Scrolling is eratic when setting limits for custom failsafe mode.

**To Reproduce**
Steps to reproduce the behavior:
1.Got to model settings for model
2. Set to FRSky D16 for protocol for internal RF(this might not matter).
3. For Failsafe mode, select custom, the click set.
4. Select a channel and try to adjust to -100 or +100 by scroll wheel - scrolling will be eratic.

**Expected behavior**
Scrolling is smoth and normal

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
EdgeTX 2.4 26b15d4d
Radiomaster TX16s

raphaelcoeffic, 2021-06-14:
Fixed in b610b22328670a8 (#193)

raphaelcoeffic, 2021-06-14:
#193 has been merged now.

closed 2021-06-14 by raphaelcoeffic