EdgeTX/edgetx #212: Pop up header?

JimB40, 2021-06-14:
**Describe the bug**
[image]
Header information is redundant and wrong color sheme

**Expected behavior**
[image]

a. Scratch pop-up header
b. Adjust color and change header text to Call-To-Action type

**General information**
- EdgeTX 77b5a3a
- RM TX16S

pfeerick, 2021-11-09:
Right, I'm in a position to change the colours on this... I just need to know what theme ids to use... 

i.e. This is using `COLOR_THEME_SECONDARY1` for the titlebar background, and `COLOR_THEME_PRIMARY1` for the titlebar text. 
[image]

For the actual text in that titlebar, I would suggest "Select mode" as this is consistent with the B&W radios.

pfeerick, 2021-11-09:
I.e. if I instead use `COLOR_THEME_PRIMARY2` for the titlebar text... it seems like that combination is readable across several themes?
[image]
[image]
[image]

JimB40, 2021-11-09:
@pfeerick is this reusable piece of code?
I think I saw som pop-up lists without header.

pfeerick, 2021-11-09:
It is the same menu, but without a titlebar set. All I need to know is what color theme parameter to use for the titlebar and what colour theme parameter to use for the titlebar text.

JimB40, 2021-11-11:
So:
1. Pop-up background is SECONDARY3
[image]

not much visible in above examples (only as 1 pixel lines between selectable list items)
Once pop-up has options on the left it's better visible
[image]
Not 100% sure but it seems wrong color is used now.

2. Not focused list item
- background - PRIMARY2
- text color - PRIMARY3

3. Focused list item
- background - FOCUS
- text color - PRIMARY2

4. Header
- background - SECONDARY1
- text color - PRIMARY2

5. Pop-up frame (frame all around pop up) SECONDARY1

pfeerick, 2021-11-11:
Sounds good, I'll try and look at that tomorrow :)

pfeerick, 2021-11-12:
Ok, so things are not quite how they seem... :grin: Don't be confused with thinking things like the background for the menu is visible on the side... that is the toolbar background... which coincidentally is SECONDARY3... ;) And it would probably be a good idea to compare with current version of the firmware, as things like the lines between entries are not present any more ;)

Scroll maker is set to PRIMARY3. 

[image]

[image]

[image]

Using your UI testing theme:

[image]

[image]

[image]

JimB40, 2021-11-12:
Looks okay Peter. 
What I'd change is that scroll marker is lying on background not on list items obscuring them.
Technically what I do in TBS AL is that I'm not changing width of list (not to hassle with window size) but I draw rectangle underneath scroll marker. Rectangle height = list height & witdh = 1-2px more than scroll marker, color = window background color. So rectangle is clipping right part of list but gives right background color and nice 1-2px whitespace / separation in background color from list items.

Regarding 'Toolbar' it ma have own color but if not defined should be transparent (window background color)

kevinkoenig, 2021-11-12:
Peter,
The lines not being there between entries looks like something I was playing with a while ago.  I thought I abandoned that, but you might find that this was a change I inadvertently made.

Kevin

From: Peter Feerick
Sent: Friday, November 12, 2021 6:24 AM
To: EdgeTX/edgetx
Cc: Subscribed
Subject: Re: [EdgeTX/edgetx] Pop up header? (#212)

Ok, so things are not quite how they seem... 😁 Don't be confused with thinking things like the background for the menu is visible on the side... that is the toolbar background... which coincidentally is SECONDARY3... ;) And it would probably be a good idea to compare with current version of the firmware, as things like the lines between entries are not present any more ;)
Scroll maker is set to PRIMARY3.

Using your UI testing theme:

pfeerick, 2021-11-13:
I'm not entirely sure on that... I don't actually think they were supposed to be there in the first place, as I noticed there are calls to a... I think it was callled removeLines...  function over on the ETX side... so the lines probably shouldn't be there... But I think when you added the ability to add separators it was also changed then also. But for the better really, as if you look at any other menu ui, the only lines are those separating groups of entries, not each individual entry.

JimB40, 2021-11-13:
This aestethic more than UX guys.
So as Bono said "With or without you" .... just be consistent :)

pfeerick, 2021-11-18:
Resolved via 38308f6bef242673a0d4d0b094d51f9bbe5784bd

closed 2021-11-18 by pfeerick