EdgeTX/edgetx #213: TOOLS penultimate item bug

JimB40, 2021-06-14:
**Describe the bug**
https://youtu.be/9ec3mXG5VVo

**General information**
- EdgeTX 77b5a3a
- RM TX16S

raphaelcoeffic, 2021-06-15:
@JimB40 I think this is actually the same issue as #114. If you have another LUA TOOL at that position, you will see that it does not happen. However, on my TX16S, the TBS Agent Lite is doing this, no matter at which position it is.

raphaelcoeffic, 2021-06-15:
Confirmed, this is a duplicate of #114.

pfeerick, 2021-06-15:
Oh, how interesting... I can actually reproduce this now... as I'm playing with a 'hello world' lua script, and since it appears in the second last row, it suffers this problem. 

It seems to be a two-fold issue
1. Regardless of what script you run, they almost all flash up the last screen from the last lua script run, before their own UI is drawn.
2. Any lua script in that last slot seems to fail to run... i.e my hello world script will run perfectly if run directly via the sd browser, but not via the tools page.
3. But, and this is the interesting bit, if I run it directly, and THEN come back to the tools page, and run it from there, it seems to launch perfectly...

JimB40, 2021-06-15:
So #114 is one bug and this is another. Just happened they joined in one place.
I'm close to resolve #114

raphaelcoeffic, 2021-06-15:
@JimB40 ok, let's re-open ;-)

raphaelcoeffic, 2021-06-15:
Seems to have been fixed by the latest fixes with the PATH used to load LUA files.

closed 2021-06-15 by raphaelcoeffic