EdgeTX/edgetx #214: Model name editing painfully slow

JimB40, 2021-06-14:
**Describe the bug**
https://youtu.be/F2AaCe0mnV4

**General information**
- EdgeTX 77b5a3a
- RM TX16S

pfeerick, 2021-06-15:
Can you try again with the current head build - https://github.com/EdgeTX/edgetx/commit/802127852e401633512dcf75f5ee5c1d6eac7b41 or https://github.com/EdgeTX/edgetx/runs/2822429567 - as it seems to be gone now.

raphaelcoeffic, 2021-06-15:
@JimB40 can you please re-test so we know if we can close it?

JimB40, 2021-06-15:
Still slow with - 8021278

raphaelcoeffic, 2021-06-15:
@JimB40 this is really strange, it works perfectly for me! (just tried on T18, I'll re-try on TX16S)

pfeerick, 2021-06-15:
I'm still on https://github.com/EdgeTX/edgetx/commit/802127852e401633512dcf75f5ee5c1d6eac7b41 on my TX16S, and I don't have the problem shown in the video... character scroll speed is just fine for me... so this really is weird... 

Video will be viewable in a minute or two: https://youtu.be/OduMayUbtZk

oryfoxer, 2021-06-15:
Can confirm, just updated to nightly and model naming is slow on TX16S

JimB40, 2021-06-15:
2 vs 2 :)

raphaelcoeffic, 2021-06-15:
@JimB40 have you tried without any LUA widgets installed? That might be related.

JimB40, 2021-06-15:
Good track. Only it's about widgets being set. Not only lua.
I have one screen with this set
[image]
None of them is LUA.
Once deleted Model name edit is fast.

JimB40, 2021-06-15:
Checked again deleting one by one. It's ModelBmp widget that slows it down.

pfeerick, 2021-06-15:
Oh, nice catch! I can repo that. It's a fluke I didn't repo it before, as I usually have ModelBmp enabled, but haven't set it up again for the models I'm testing on.

raphaelcoeffic, 2021-06-15:
> Checked again deleting one by one. It's ModelBmp widget that slows it down.

Thx! This is something I can deal with. The refresh rate is probably way to fast!

commit a0090f4e (2021-06-16): [Color] Prevent ModelBmp from reloading the bitmap when not needed

Resolves #214
closed 2021-06-16 by raphaelcoeffic