EdgeTX/edgetx #216: Change USB screen to preserve radio battery

JimB40, 2021-06-14:
Existing screen

[image]

Proposed
[image]
[image]

Don't know if OS is working while in USB mode. If so second screen provide possibility to check visually radio battery level

raphaelcoeffic, 2021-06-14:
@JimB40 the icon is not bad, indeed! Do you have it separately?

rotorman, 2021-06-14:
Love the proposed new logo - much more appropriate than the old one in a USB-stick form.
I would add text "USB" to make it clear.

pfeerick, 2021-06-14:
I was going to suggest something similar when 2.4.0 was released, namely that the USB modes be more aligned with what the B&W radios can do... You can see the battery level, time, stick and switch widgets etc. 

I.e. in USB Storage mode you could just show an icon like with the joystick mode, and lock touchscreen input and buttons/roller so you can't go into menus.

raphaelcoeffic, 2021-06-15:
Oh ok! For my part I thought this would be better for the mass forage mode, as you should believe you can use the menus of anything else while in that mode. For the 2 others, you can still use the radio normally.

pfeerick, 2021-06-15:
Proposition 2 can be another 'thing' for 2.4.1... just keep the prettier icon for now, and probably dim the screen a bit...

pfeerick, 2023-12-06:
@JimB40 Do you still have the source graphic for this and can it be freely used? Otherwise, this will be moved to 2.11.

richardclli, 2023-12-06:
Well, changing the graphics cannot preserve the battery on a LCD display.  If you really want to preserve the battery, then you need to lower the brightness instead.

pfeerick, 2023-12-06:
Indeed... as it is the backlight brightness that determines the power draw, not the content of the display. I was just focusing on the graphic though for this... if Robert still has it, great as it looks nicer, if not, close and move on.

richardclli, 2023-12-18:
But power will be different if OLED display is used instead.  However, no color radio is using an OLED display right now.

closed 2024-04-05 by pfeerick