EdgeTX/edgetx #222: LUA: on colorlcd radios widget's background function is never called

yaapu, 2021-06-15:
last tested on commit f18a36c27

I added a couple TRACEs and LuaWidget::background() is never called

raphaelcoeffic, 2021-06-15:
> I added a couple TRACEs and LuaWidget::background() is never called

The code for widgets has changed quite a bit compared to 2.3. The question is now, how should it be mapped?

ATM, the UI loop looks like this:
- `checkEvents()`: called for each loop and each UI component. If `invalidate()`is called on a UI component, this will trigger a re-draw in step 2.
- `paint()` is called for all visible components.

The LUA widgets are mapped to this logic:
- every time the update period has passed, `checkEvents()` will trigger a refresh (with `invalidate()`).
- `LuaWidget::refresh()` is called, whether or not the widget is visible. Clipping will take care of not drawing anything if the widget draws off-screen.

What I could imagine would be to call `background()` instead of `refresh()` when the widget is not visible.

Would that be ok?

yaapu, 2021-06-15:
> The LUA widgets are mapped to this logic:
>
> every time the update period has passed, checkEvents() will trigger a refresh (with invalidate()).
> LuaWidget::refresh() is called, whether or not the [...]

yes makes sense to me

raphaelcoeffic, 2021-06-15:
> yes makes sense to me

I just re-checked in the code: this won't be possible, as `paint()` is not called if the component is not visible, and `checkEvents()` cannot know if something is visible or not....

So the only thing that we can do is calling `background()` from `checkEvents()` at a fixed frequency (somehow, depends on main UI loop speed, which can be delayed by slow drawing, or anything in that direction).

yaapu, 2021-06-15:
> I just re-checked in the code: this won't be possible, as `paint()` is not called if the component is not visible, and `checkEvents()` cannot know if something is visible or not....
>
> So the only [...]

we've always had background() delayed by foreground painting, what would a good starting point be, 50Hz? right now my widget which is kind of a heavy weight runs at 10Hz in foreground, also you'd have to share the background call for all bg running widgets, so each widget is running in the main GUI thread?

raphaelcoeffic, 2021-06-15:
> we've always had background() delayed by foreground painting, what would a good starting point be, 50Hz? right now my widget which is kind of a heavy weight runs at 10Hz in foreground, also you'd [...]

This would be widget by widget, so that we possibly get a statistical distribution.

lshems, 2021-06-17:
I think it is very bad practice to get the backgroud processes being delayed by drawing. Therefore I don't like the call for the background that is needed when the widget is in the foreground, which is what is documented on OpenTX for widgets to be the case.

The background could run at an entirely different pace (it is OpenTX/EdgeTX that manages the cycled function calls!).

But then I would run into issues when doing things in background that I need to signal to foreground. Extra complexity.

Therefore I would really like another concept, where ONE script is loaded and executed and the script itself determines the pace at which it does what. So the script that is loaded has to start an endless loop itself, in which it periodically does things.

I hope what I write is understandable.

P.S. this would also limit the need to manage speed. Set it at maximum and equal to all, and if you want to slow down one script in favor of the other, manage the pace yourself in your looping. You can even use semaphores between widgets, as they do share the same global environment, to orchestrate widgets among each other.

raphaelcoeffic, 2021-06-17:
> I think it is very bad practice to get the backgroud processes being delayed by drawing. Therefore I don't like the call for the background that is needed when the widget is in the foreground, which is what is documented on OpenTX for widgets to be the case.

it’s not like we would have real multi-threaded processing ;-) this is still running on a quite constrained MCU, don’t forget it. But we might get some improvement once we switch to FreeRTOS. We will see.

lshems, 2021-06-18:
By the way the behaviour noted in the first comments is according to OpenTX widget specs. You should call background yourself if you want it to run while the widget is on screen.

jfrickmann, 2021-06-19:
I have spent a couple of nights studying the colorlcd gui code in the hope that I could say something clever here. But so far I have not yet managed to grasp how it works in details 😕
But for telemetry scripts, background() and run(event) functions are driven from luaTask(). And this is what I modified in my PR to make it preemptible (https://github.com/opentx/opentx/pull/8194). How about we make the Lua widgets share the Lua state with the other scripts, and drive the background functions from luaTask? It would have the added value that function scripts and widgets could share global variables. In SoarOTX, I use a function script as a kind of library with shared functionality plus some sensor data management. This would be very useful to also have with widgets.

yaapu, 2021-06-19:
> P.S. this would probably mean that background will be called also when the widget is visible, just like it works for telemetry scripts. Not a problem IMHO

not a real issue but widgets will have to detect this behaviour at runtime to cope with OpenTX 2.3 background() not called when in foreground

raphaelcoeffic, 2021-06-19:
@yaapu it looks like there is a simple trick to get the background function executed only when not visible: check if `paint()` has been called after the widget has been `invalidate()`-ed. Have a look at the small PR #271.

jfrickmann, 2021-06-19:
Yes, we can probably fairly easily prevent background() from running when the widget is visible.
But how about moving the widgets over to share Lua state with the other scripts and having calls to background() be driven from luaTask() ?

raphaelcoeffic, 2021-06-19:
> Yes, we can probably fairly easily prevent background() from running when the widget is visible.
> But how about moving the widgets over to share Lua state with the other scripts and having calls to background() be driven from luaTask() ?

Either way there will be mutexes. I need to think of a smart way of doing it. Did you test the new PR for the background function?

PR #271 (2021-06-22): Call LuaWidget background() if widget is not painted
Resolves #222
commit 9ecccef3 (2021-06-22): Call LuaWidget background() if widget is not painted

Resolves #222
closed 2021-06-22 by raphaelcoeffic