EdgeTX/edgetx #223: Trainer signal input woes plus Special Function Trainer Chans

rotorman, 2021-06-15:
Tested with ba7ce9e9 on RM TX16S.

For the following tests, I set special function 1 using switch SC up, to Function: Trainer, Value: Sticks and enabled it. Set switch to up.

Test 1 - cPPM:
When setting MODEL -> MODEL SETUP -> Trainer Mode to Master/Jack and entering a cPPM signal via trainer jack (from Futaba Tx).
No change can be observed in SYSTEM->Trainer nor in Channel Monitor.

Test 2 - SBUS
SYSTEM->HARDWARE->Serial port is set to SBUS Trainer and MODEL -> MODEL SETUP -> Trainer Mode is set to Master/Serial.
When hooking TX16S AUX1 up to a SBUS outputting device, can hear Trainer Signal recovered and by disconnecting Trainer Signal Lost sounds. SYSTEM->TRAINER shows the values of **first 4** channels correctly.  When having SC up, I also see in Channel Monitor output from channels 1 to 4, but not higher channels. By looking with debugger, all 16 SBUS channels are input and parsed correctly for _ppmInput_:
https://github.com/EdgeTX/edgetx/blob/ba7ce9e911c5628f0a0fdceb9aeffdda33080139/radio/src/sbus.cpp#L87

I wonder if  `FUNCTION_TRAINER_CHANNELS` is defined correctly here at all:
https://github.com/EdgeTX/edgetx/blob/ba7ce9e911c5628f0a0fdceb9aeffdda33080139/radio/src/opentx.h#L842
I believe it should be Sticks = 0, Rud=1, Ele=2, Thr=3, Ail=4, Chans=5, thus:
`FUNCTION_TRAINER_CHANNELS = FUNCTION_TRAINER_STICK1 + NUM_STICKS + 1` (+1 is missing) ?

rotorman, 2021-06-15:
Manually adding a mixer for TR5..TR16 works, but naturally in this case the Trainer special function (SC in above example) has no relevance.
Tested with OTX 2.3.11 for comparison, there it works as expected, but have not been able to identify the culprit line in src.

rotorman, 2021-06-15:
I made a comparison video with ETX vs. OTX to show the behaviour. Second radio is hooked to TX16S via trainer cable/cPPM:

EdgeTX: https://www.youtube.com/watch?v=huZDmEKjn5s
OpenTX: https://www.youtube.com/watch?v=Rq1w5mgnVok

raphaelcoeffic, 2021-06-16:
Fixed by 6b59660225169

closed 2021-06-16 by raphaelcoeffic