EdgeTX/edgetx #227: some model functions still causing problems in LUA

lshems, 2021-06-16:
Just loaded the preleased and I'm impressed (8cc5e77).

I'm still struggling to find my resources and info though.
My scripts run mostly fine, except for those using a lot of model functions. I cannot really debug, since I have to load the code on the radio and have no LUA debug window there. 

So is there a way to debug lua scripts?
Secondly, where do I find the info on how to use the keys in a full screen widget? or is it not added yet? That would help debugging also, since I can then at least use the widgets interactively and add some error catching perhaps in lua itself.

Thanks again for all effort.

raphaelcoeffic, 2021-06-16:
@lshems may I help you with something? The easiest for me so far has been to use the standalone simulator called „simu” in a terminal. Please PM me on discord in case you need to give me some code to test and check.

lshems, 2021-06-16:
Code checking is really difficult for someone else. Its 400+ files of lua. 
I created an entire library of stuff to make things easy. I don't have an issue with sharing, but it's really not helpful. I am willing to put my own effort in, and I have the simu running on windows all the time for opentx. But where to find it for Edge?

Discussion on discord then? I can create a wiki page on how to do things once I figured it all out.

pfeerick, 2021-06-17:
Simu on windows is probably going to be a pain since we can't currently build companion24, simulator24 or simu on Windows... so your best bet at present is to build and run `simu` on Linux. You can build it on WSL2, but then the inability to easily run GUI stuff gets you ... although that *may* be resolved in the next week or after the June 24 Windows announcement, since among other things, the X11 GUI stuff  - WSLg "is going to be generally available alongside the upcoming release of Windows" ... 

It is entirely possible though, although like pulling teeth, to run simu right now on Windows via WSL2... 
Follow [these instructions](https://medium.com/javarevisited/using-wsl-2-with-x-server-linux-on-windows-a372263533c3) to get the X11 server up and running. It's up to you if you make it so stuff runs as startup, or if you just create the vcxsrv shortcut, and vbs file, and then just run them manually when you want the GUI. 

Build and make the simulator - it's the `simu` target, rather than the `firmware target`... I use the docker build environment, so for me it's something like (when the present working directory is the root of the edgetx repo):
```
docker run -it --rm -v $(pwd):/src ghcr.io/edgetx/edgetx-dev bash
cd /src/build (mkdir /src/build and cd /src/build if it's missing) 
cmake -DPCB=X10 -DPCBREV=TX16S -DLUA=YES -DCMAKE_BUILD_TYPE=Release ..
make -j4 simu
exit
./simu
```

You'll probably run into the libfox shared library error, so run `sudo apt install libfox-1.6-0` to fix that. 

[image]

raphaelcoeffic, 2021-06-17:
@lshems you might be interested in re-testing with a newer version, as there are some fixes in d10563d08b2 concerning the labels used in such model functions.

lshems, 2021-06-17:
Yeah, saw that. Nice. Will do.

lshems, 2021-06-17:
OK, no big improvements after the update. Im writing a small menu page to show all possible key-values for all possible screen objects (inputs, mixers, outputs, LS, SF I have up and running on the sim in OpenTX, just find the time to test on the radio.)
[image]

lshems, 2021-06-17:
> Simu on windows is probably going to be a pain since we can't currently build companion24, simulator24 or simu on Windows... so your best bet at present is to build and run `simu` on Linux. You can build it on WSL2, but then the inability to easily run GUI stuff gets you ... although that _may_ be resolved in the next week or so after the June 24 Windows announcement, since among other things, the X11 GUI stuff - WSLg "is going to be generally available alongside the upcoming release of Windows" ...
> 
> It is entirely possible though, although like pulling teeth, to run simu right now on Windows via WSL2...
> Follow [these instructions](https://medium.com/javarevisited/using-wsl-2-with-x-server-linux-on-windows-a372263533c3) to get the X11 server up and running. It's up to you if you make it so stuff runs as startup, or if you just create the vcxsrv shortcut, and vbs file, and then just run them manually when you want the GUI.
> 
> Build and make the simulator - it's the `simu` target, rather than the `firmware target`... I use the docker build environment, so for me it's something like (when the present working directory is the root of the edgetx repo):
> 
> ```
> docker run -it --rm -v $(pwd):/src ghcr.io/edgetx/edgetx-dev bash
> cd /src/build (mkdir /src/build and cd /src/build if it's missing) 
> cmake -DPCB=X10 -DPCBREV=TX16S -DLUA=YES -DCMAKE_BUILD_TYPE=Release ..
> make -j4 simu
> exit
> ./simu
> ```
> 
> You'll probably run into the libfox shared library error, so run `sudo apt install libfox-1.6-0` to fix that.
> 
> [image]

OK, going that route now. But can someone share perhaps a precompiled simu, as it would save me the complexity to test the WSL2 also for compiling

raphaelcoeffic, 2021-06-17:
> OK, no big improvements after the update. Im writing a small menu page to show all possible key-values for all possible screen objects (inputs, mixers, outputs, LS, SF I have up and running on the sim in OpenTX, just find the time to test on the radio.)
> [image]

If you provide your test script, I can have a look what   Is not working properly.

lshems, 2021-06-17:
OK, fianlly got a working simu with lua debug. now I can work again :).

First issue: on inputs, the name attribute is empty. This causes my scripts to crash because I look for inputs with a certain name. the InputName is populated, but name is not.
[image]

lshems, 2021-06-17:
It seems a conversion issue, as I can populate it myself:
[image]
[image]
[image]

lshems, 2021-06-17:
Perhaps the other errors are also caused by conversion issues. I'll check, and will change the titel?

lshems, 2021-06-17:
Also, this means that the long standing bug is fixed, with no access to the inputname and line name as seperate parameters on inputs from LUA.

Nice

lshems, 2021-06-18:
I will close this issue, and create clean issues for what results from further testing.

So I will create a new issue for the input line names not being converted properly.

closed 2021-06-18 by lshems