EdgeTX/edgetx #228: Popup list filters need more work

pfeerick, 2021-06-17:
For want of a better word, the category/filter icons that appear on the side of lists ... 

[image]

... can make it so you can't see all the list, i.e. if you press the top one for switches, it will only show the 'positive' switch states, not any of the not states, and the second one will only show the sticks... so if you use either filter, you loose access to the not switch states, 'Tele' and 'Act' etc in this examble. Another is if you were in the global functions screen, if you hit one of the filter icons, you can't get at the ON state unless you close and reopen the list - as is see here... => https://youtu.be/Y9OvW9XCjOs?t=1440

Perhaps at minimum an 'all' option is needed so you can always get the full unfiltered list back?

pfeerick, 2021-09-15:
Another suggestion was that the GVs, LSs and a Misc or something for things like Batt, and Timers

closed 2021-11-30 by pfeerick

JimB40, 2022-10-02:
Partially done in https://github.com/EdgeTX/edgetx/pull/2435

JimB40, 2022-10-02:
- [ ] Reaarange sources' filter icons in two columns so in horizontal layout icons scrolling is not needed.
- [ ] Add top bar with variable name for clarity what is actually changed
- [ ] Add close button on topbar for UI clarity (tapping ouside the the list still works)

pfeerick, 2022-10-02:
Negative. the changes in #2435 don't relate to this issue, and it's a close issues anyway ;)

Eldenroot, 2022-10-02:
It is always frustrating to scroll over all items... nightmare, I hope it will be improved in 2.9 somehow :) some QoL improvements are welcomed!

commit 2bdf1ff6 (2021-11-30): [libopenui ] Fix enable/disable popup menu filters
Can now enable and disable filters in popup menus (i.e. source, switch)
Active filter is now also highlighted
Use page keys to scroll through filters if no touch
Resolves #228