EdgeTX/edgetx #229: Touchscreen sometimes becomes completely unresponsive

pfeerick, 2021-06-17:
Just documenting this as I've had it happen a couple times as well - no repeatable reason as yet - but it did always 'seem' to be when in model settings. Only solution was to power cycle radio. 

https://youtu.be/Y9OvW9XCjOs?t=1753

raphaelcoeffic, 2021-06-17:
@pfeerick this is going to be a rough one.... we need to find some way to trigger it, so I can check what's happening.

pfeerick, 2021-06-17:
Exactly... it happens every now and then, just enough to remind you it can do it, not enough to actually start pinning down the commonality. I don't see it being a **big** issue since touch is only complementary to the button interface, there's nothing it alone allows you to access AFAIK. It's just a matter of finding out where it's getting stuck ;)

phileville, 2021-06-19:
I also had this happen yesterday - also when I was on the model configuration screen - but I could not recreate it again and it only happened once.  Another user from FB also mentioned it. I asked him for more information.

raphaelcoeffic, 2021-06-19:
> Just documenting this as I've had it happen a couple times as well - no repeatable reason as yet - but it did always 'seem' to be when in model settings. Only solution was to power cycle radio.
> 
> https://youtu.be/Y9OvW9XCjOs?t=1753

Did you use any of the screen settings during the same session? Anything with exchanging a widget, selecting a model with Toolbar widgets?

PR #282 (2021-06-20): Fix attempt for locked touch
Thx Jakub!

Resolves #229.
closed 2021-06-20 by raphaelcoeffic