EdgeTX/edgetx #230: Panel mirroring option does nothing

pfeerick, 2021-06-17:
The Mirror option under Screen Settings -> Main view appears to do nothing. Expected behaviour was that it would mirror the layout. 

[image]

Seen in this video https://youtu.be/Y9OvW9XCjOs?t=2721 and reproduced locally on TX16S and on simu.

JimB40, 2021-06-17:
It is mirroring. Still widget visual output stay at the same UI place until you edit it.

pfeerick, 2021-06-17:
Or until you power the radio off... I just saw it had applied on a power cycle. So in other words, unlike all the other checkboxes on that screen, that one does not apply 'immediately'. So you can either power cycle the radio, or bring up the widget settings to get it to refresh the layout mirroring setting.

raphaelcoeffic, 2021-06-17:
> Or [stays the same] until you power the radio off... I just saw it had applied on a power cycle. So in other words, unlike all the other checkboxes on that screen, that one does not apply 'immediately'. So you can either power cycle the radio, or bring up the widget settings to get it to refresh the layout mirroring setting.

Thx, no problem, that should be easy to fix.

commit 64e2b25a (2021-06-17): Fixed layout update when "Mirror" option is changed

Resolves #230
closed 2021-06-17 by raphaelcoeffic