EdgeTX/edgetx #231: V2 value on the logical switches setup screen cannot be set negative

eshifri, 2021-06-17:
**To Reproduce**
Steps to reproduce the behavior:
1. Go to the model setup logical switches page.
2. Click on the logical switch and select Edit.
3. Select V1 e.g. "Thr"
4. Click on V2 edit. The available range is [0, 255].

**Expected behavior**
The range should be [-100, 100]

**General information**
 - EdgeTX version 2.0.0 RC1
 - Radiomaster TX16S

eshifri, 2021-06-17:
I suspect line 171 in model_logical_switches.cpp instead of 
v2Edit = new NumberEdit(logicalSwitchOneWindow, grid.getFieldSlot(), 0, MAX_LS_DELAY, GET_SET_DEFAULT(cs->v2));
should be 
v2Edit = new NumberEdit(logicalSwitchOneWindow, grid.getFieldSlot(), v2_min, v2_max, GET_SET_DEFAULT(cs->v2));

Sorry, do not have time to try right now.

raphaelcoeffic, 2021-06-17:
> I suspect line 171 in model_logical_switches.cpp instead of
> v2Edit = new NumberEdit(logicalSwitchOneWindow, grid.getFieldSlot(), 0, MAX_LS_DELAY, GET_SET_DEFAULT(cs->v2));
> should be
> v2Edit = new [...]

Thx, yes, makes sense to me, I’ll try it out.

raphaelcoeffic, 2021-06-17:
@eshifri works perfect, thx!

commit 87610b84 (2021-06-17): Fixed logical switch V2 edit value range

Resolves #231
closed 2021-06-17 by raphaelcoeffic