EdgeTX/edgetx #236: Splash screen timeout broken on X9D+

Master92, 2021-06-17:
**Description**
I just built the latest commit of the 2.4 branch (d02298c61) on a brand new virtual Ubuntu 20.04 machine for my Taranis X9D+ with the option HELI disabled and everything else on default.

Whenever I try to change the timeout value for the initial splash screen, I can only choose 3 or 4 seconds and can not select anything else or switch it off at all.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Radio Setup'
2. Scroll down to 'Splash Screen'
3. Try to change it to something else than 3 or 4s
4. See error

**Expected behavior**
In OpenTX 2.1.12 I have the possibility to choose between:

- ----- (Off)
- 2s
- 3s
- 4s
- 6s
- 8s
- 10s
- 15s

I expected to have the same choices in the current EdgeTX version but it only cycles between 3 and 4s.

**General information**
 - EdgeTX version 2.4
 - FrSky Taranis X9D+
 - Companion version - doesn't matter, radio specific issue

pfeerick, 2021-06-17:
I can repro this, on the X9D+ and T-Lite on different commits from the last 2-3 days.

RohanClegg, 2021-06-17:
I'm having this exact same issue on both my Taranis X7 and Jumper T-Lite. There are no other options for splash screen delay other than 3 and 4 seconds, running the latest build of EdgeTX.

commit fd02387c (2021-06-19): Restored proper data format for splashMode

Resolves #236
PR #268 (2021-06-20): Restored proper data format for splashMode
Resolves #236
closed 2021-06-20 by raphaelcoeffic