EdgeTX/edgetx #237: Power on / Splash screen delay options

pfeerick, 2021-06-17:
In looking at #236, I noticed that the TX16S doesn't have power on or splash screen delay settings... I don't know if the former is possible, but some people would probably like the latter.

raphaelcoeffic, 2021-06-17:
> In looking at #236, I noticed that the TX16S doesn't have power on or splash screen delay settings... I don't know if the former is possible, but some people would probably like the latter.

At the moment, there is no really delay coded anywhere on the TX16S. I’m still trying to figure out why it takes so long until the MCU takes control on boot. This is probably some hardware issue (same on X10, by the way).

aquaaddict, 2021-07-10:
I also am not seeing a Splash Screen Delay option on the Tx16S, and agree it would be great to have one please

raphaelcoeffic, 2021-07-11:
> I also am not seeing a Splash Screen Delay option on the Tx16S, and agree it would be great to have one please

Just to scope the goals: would you configure it to be longer than what it is now? At the moment, there are no delays at all.

aquaaddict, 2021-07-12:
Yes, a configurable delay to show the splash screen for longer, eg configurable number of seconds, typically values in the range 3 to 10 seconds. The splash screen image is customisable but that because of little use if the image quickly flashes past.
Tha ks!

ikherty, 2022-01-06:
EdgeTx 2.6 RC2
Taranis X7

taranis x7 does not respond to changes in parameters "power on delay" and "splash screen". I changed to 0s in both parameters, but still the transmitter turns on for a long time. "Power off delay" works correctly.

pfeerick, 2023-11-04:
Resolved by #https://github.com/EdgeTX/edgetx/pull/3451

closed 2023-11-04 by pfeerick