EdgeTX/edgetx #239: Black screen with a arrow after splash screen while booting

sprrrry, 2021-06-17:
I am getting a black screen with a arrow on the bottom right after the splash screen while booting every time. This is happening in all builds after the modelbmp cache fix.

[modelandradio.zip](https://github.com/EdgeTX/edgetx/files/6669735/modelandradio.zip)
https://user-images.githubusercontent.com/38971821/122387232-47afb280-cf8c-11eb-8e0c-943f5a7f7568.mp4

pfeerick, 2021-06-18:
Just to confirm, I saw it flash up with the latest HEAD (https://github.com/EdgeTX/edgetx/commit/952bc12b0429ea02f6779bb2088fdc6a0dd93532) ... no where near as long as this though...

sprrrry, 2021-06-18:
Just checked, the time for which the black screen is active is proportional to the size of the model image. When I don't use any model image there is no such screen and it is like before the modelbmp fix was done and when using a 192x108 image (which I guess you are using too) the black screen is very short. In the video I used a 480x272 image and so the time is longer. I think the black screen is just the image caching.

raphaelcoeffic, 2021-06-18:
@sprrrry thx for the test work! This is definitely something I can use!

pfeerick, 2022-09-08:
I've just re-tested for this issue, as I suspected it had been fixed in a PR for 2.7 and it is certainly no longer present in 155a854131d142a1740dd3a57b40a94e98dcad6a. I've not seen this artifact for quite a long time now, and certainly not with the easiest method of reproducing (use a stupidly large model image). Now, all that happens is the splash screen shows for longer while the model image is being loaded.

closed 2022-09-08 by pfeerick