EdgeTX/edgetx #241: 6pos switch calibration tales multiple iterations

eshifri, 2021-06-17:
**Describe the bug**
It takes several swipes through the 6pos switch to get it calibrated

**To Reproduce**
1. Go to HW calibration, calibrate central positions.
2. Proceed to range/end point calibration.
3. Proceed to 6 pos switch calibration.
4. It takes about 10 - 15 runs through all buttons to calibrate the switch.
5. There is a very long delay between pressing a button and the change of the number on the screen.

**Expected behavior**
One pass through the buttons should be sufficient. It works that way in OTX 2.3.11

**General information**
 - EdgeTX 2.4.0 RC1
 - Hardware RM TX16S

raphaelcoeffic, 2021-06-17:
@eshifri the delay might be due to an issue with ModelBmp which has been fixed in between. You could re-test with the latest nightly build.

eshifri, 2021-06-17:
Yes, working reasonably good now!

raphaelcoeffic, 2021-06-18:
> Yes, working reasonably good now!

Ok, but I guess the cumbersome calibration is still an issue, right?

eshifri, 2021-06-18:
It is also better now: I guess slow reaction of the screen was part of the problem. I did not have enough patiecne to wait to the screen changes.
It is still not as fast as in OTX 2.3, but two-three passes through the switches are not too bead.

pfeerick, 2021-06-18:
I noticed right from the start when doing the initial calibration that the 6POS were slow... didn't give it much thought... since they're dopey to start with (buttons faking a pot via a dedicated IC, meaning they're not controllable by the TX :( )... maybe if there is generally some way to suspend any other lua stuff whilst in the calibration screen that would be good... it's not like any widgets needs to be updating ;)

JimB40, 2021-11-11:
Still valid?
I was calibrating TX16S lastly. Have some little lag, but sequentially pressing 1-2-3-4-5-6-1 does the job well.

eshifri, 2021-11-11:
Last time I tried it, it took more than one pass.

pfeerick, 2021-11-11:
*Seems* to work better if you treat it like a pot/slider, since that is what it is emulating... so press 3 or 4 when told to center everything, and then run from 1-> 6, and then 6 -> 1. But even so, I think calibration screen lags in comparision to normal UI response.

eshifri, 2021-11-11:
That is the procedure I'm doing. Still it takes 2 - 3 passes to get it calibrated while normal pot is done in one pass.
Not terrible, but there is space for improvement. :-)

eshifri, 2021-11-21:
I think it works good enough now.

closed 2021-11-21 by eshifri