EdgeTX/edgetx #242: errors in creating Ubuntu compile toolchain using instructions on wiki

lshems, 2021-06-17:
The instruction says I should not see errors. Unfortunately::

[image]

lshems, 2021-06-17:
[image]

lshems, 2021-06-17:
Blocked here. Installed cmake, but then its done. No clue how to continue.

rotorman, 2021-06-17:
This script is for bare-metal or virtual machine Ubuntu 20.04 installation, not in containerized. Apparently the Windows Ubuntu WSL image is missing some package sources.

lshems, 2021-06-17:
Yeah, ok. It's wsl2 by the way. But ok.

rotorman, 2021-06-17:
@lshems can we close this issue?

lshems, 2021-06-17:
Yes and no. I managed to get a radio firmware compiled.

For companion it stalks on the credits target. 

Not convinced it could not work

lshems, 2021-06-17:
make[3]: *** No rule to make target '../CREDITS.txt', needed by 'companion/src/qrc_companion.cpp'.  Stop.
make[2]: *** [CMakeFiles/Makefile2:2411: companion/src/CMakeFiles/common.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:2392: companion/src/CMakeFiles/companion24.dir/rule] Error 2
make: *** [Makefile:1044: companion24] Error 2

when trying to compile companion

rotorman, 2021-06-17:
Yep, I already pinged @pfeerick about this (he cleaned some OpenTX leftovers today, but companion and simulator binaries depend on the CREDITS.txt). Just create an empty CREDITS.txt in the root of EdgeTX folder and you will be good to go (just tested).

lshems, 2021-06-17:
What is root?

lshems, 2021-06-17:
build-output

rotorman, 2021-06-17:
The uppermost dir, where radio, companion, tools and so on folders are.

rotorman, 2021-06-17:
I just re-created an empty CREDITS.txt in the git repo, so that now if you git pull, you should be good to go as well.

lshems, 2021-06-17:
Well, got the simu started. Downloading sdcard content now.

Can I get a wiki access to create a page on how to get it to work? it will start a nice simu window in windows, hopefully :)

rotorman, 2021-06-17:
> Can I get a wiki access to create a page on how to get it to work? it will start a nice simu window in windows, hopefully :)

If you send me the markdown and images (e.g. via Discord private message as zip), I can integrate it. If this is not a viable option for you, plz ping @raphaelcoeffic , e.g. via Discord about access rights.

lshems, 2021-06-17:
Can we get a default profile so we don't need companion? I need to create a profile first now in companion. Unnecessary step I would think (I hate companion).

rotorman, 2021-06-17:
Indeed this is a valid point you are making - in order to run full simulator, need to start companion once first and make a profile.
You are very welcome to propose a change in the form of a PR (wink, wink).

lshems, 2021-06-17:
[image]

lshems, 2021-06-17:
OK, someone give me some wiki access and ill create an install guide with basic usage instructions

lshems, 2021-06-17:
> Indeed this is a valid point you are making - in order to run full simulator, need to start companion once first and make a profile.
> You are very welcome to propose a change in the form of a PR (wink, wink).

I can't change one letter of Ccode without crashing it.sorry.

pfeerick, 2021-06-17:
Whilst `simu` is not as fancy, it doesn't need a radio profile as it is completely self-contained... so much so you can just leave it in the root of your sd card and it will just work from there. :wink: (i.e. `make -j4 simu`)

pfeerick, 2021-06-17:
Thanks @rotorman - that was going to be my first plan of attack - a dummy `credits.txt` until I can remove the dependencies. Sorry about that @lshems! :innocent:

lshems, 2021-06-17:
No, it's fine. I've at least proven that we can run a windows ubuntu install with xserver to run the simu. It would really be nice if the simu could be put in the distro as well, even if only one per redio output type: xd9+, xlite-pro and t16 or something like that.

lshems, 2021-06-17:
> Whilst `simu` is not as fancy, it doesn't need a radio profile as it is completely self-contained... so much so you can just leave it in the root of your sd card and it will just work from there. 😉 (i.e. `make -j4 simu`)

No. It asks for a profile and if it doesn't find it, refuses to start.

lshems, 2021-06-17:
I only need simu. Companion by the way wont start a simu on my install??

lshems, 2021-06-17:
ANyway, I will leave this open until I can update the WIKI.

pfeerick, 2021-06-17:
`simu != simulator24` :wink: 

This is simu. It's not as friendly as simulator, as it won't ask for profile, radio, nothing... if it runs from the root of a SD card, it will use it, else it will just behave like the version of the firmware and radio it was compiled for. It's radio specific, so you can't just witch to another radio like with the full simulator. 

btw, there will be no end of issues with companion, as it's basically only been kept up to date with OTX 2.3, I don't think any OTX2.4 specific changes have been made, let alone any ETX changes. 

[image]

lshems, 2021-06-18:
Ah super. This will help inme setting up a wiki for testers. KISS

lshems, 2021-06-18:
https://github.com/EdgeTX/edgetx/wiki/Setting-up-Ubuntu-20.04-running-in-a-Windows-Subsystem-for-Linux

lshems, 2021-06-18:
OK, how can we add a SIMU for X9D+ somewhere so people can fetch it from the Ubuntu and try to run it? Can we add it somewhere on GitHub. I'm a total noob on github, I just want users to be able to fetch this basic sim that can run without SDCARD content on a basic ubuntu on a Windows subsystem without anything else done yet. So without even preparing the compile toolchain.

rotorman, 2021-06-18:
So you want a static present day build binary? For static files one could create a subfolder \files under Wiki and link it from there, although a static file will be pretty outdated after a while, looking at the pace EdgeTX is being developed.

lshems, 2021-06-18:
Ok, such a folder would make sense for screenshots to use in the wiki as well, no?

Can you add it? Then I'll probably will be able to figure out how to use it.

The idea is indeed to have a static file there for some radio types, just so people can test Lua scripts for instance without having to compile.

Off course, when there is a stable release, we could provide just the latest version there.

rotorman, 2021-06-18:
For screenshots the `\media` subfolder in Wiki is perfect.

Although, I do not think serving binaries from Wiki is the way to go in the long run, I added for now for you `\files\simu\simu_x9dplus_ubuntu` so that you can continue. The full URL is:
`https://raw.githubusercontent.com/wiki/EdgeTX/edgetx/files/simu/simu_x9dplus_ubuntu`

Best you git clone the wiki locally, then you can do all the work first locally before you commit/push. If you are running Windows and like GUI based tool, try e.g. TortoiseGIT.

I guess, this issue can be closed?

lshems, 2021-06-18:
Why push. I've managed to get it running on the wsl2. I'm working on the install script to see where it blocks.

Would be nice if I can get it fixed do people on windows machines can easily create a compile street from which they can also run?

Or am I doing things the wrong way. Then tell me how it can be done easier.

Compiling AND running simulator and companion on a Windows machine.

rotorman, 2021-06-18:
Building simu under WSL2/Ubuntu does not result in native Windows binaries at the end (they still will be Ubuntu binaries, just running encapsulated in WSL2/Ubuntu corset). The binaries are no different, than the ones when running inside VM with Ubuntu.
The goal IMHO is to get the radio simulator library to build natively under Win for users of that OS, as is the case with OTX 2.3.
EdgeTX Companion and Simulator binaries (well basically the same as OTX 2.4) already build just fine, but the radio fw libs are giving still head-ache.

lshems, 2021-06-18:
You are not getting the point. I don't care, and neither do a lot of others I think, where the simulator or the companion is compiled, nor if it is run natively in windows or in a X-window.

For now that is.

People just want to be able to run the simulator NOW on a windows machine, with as little complications as possible, and without waiting for the problems on windows to be solved.

I found that it is perfectly possible to run Ubuntu in WSL2, and run the companion, simulator and simu simple simulator, effectively giving me the same tools as if they where compiled on windows. I can comile my own radio types, my own compile options, without a lot of hassle.

With the explorer.exe command from Linux an explorer window opens with which I can even access the files from the Ubuntu environment safely. And I can from companion or the simulator use the SDcard images and otx files on the windows machine, just use the correct mountpoint /mnt

Anyway, I found the problem with the install files: on the windows UBUNTU there is no i386 support. I made adding that support the last step on the wiki for the WSL2 install of UBUNTU.

ALL WORKS, so closing the issue. :)

Thanks to all who have contributed.

closed 2021-06-18 by lshems

pfeerick, 2021-06-19:
Also keep in mind what I said before about WSLg as it will probably be even easier to run simu under WSL2 after the 24th depending what comes out of the Microsoft announcement then.