EdgeTX/edgetx #243: The necessary lua probably work well for me. When upgrading to EDGE, it did not load the widget settings from OpenTX on the screen at startup. I need to set them again. The internal ISRM is not displayed. This does not work with the ACCESS protocol. It does not contain Czech, which does not matter. But I understand English ... User color themes on the display would also be good. Otherwise EDGE is very nicely processed and I like it. Good work. Good luck.

mkdaw, 2021-06-17:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX version (e.g. 2.4.0 or nightly and #commit number)
 - Hardware (e.g. Frsky X9D+ or RM TX12S)

raphaelcoeffic, 2021-06-18:
@mkdaw we probably need to state it better: the lost widget settings / screen settings is intentional, as it is not compatible to 2.3. We are using typed options, whereby 2.3 uses an implicit semantic, assuming the code for these widgets does not change…. So we had to wipe it so that it does not cause trouble.

pfeerick, 2021-06-18:
So another known limitation for the notes? i.e.
* Panel / widget settings cannot be preserved in the conversion from OTX 2.3.x => ETX 2.4.0 due to incompatibility in how the settings sere stored. 

Is there a better way to say in plain language how converting the OTX 2.3.x => implicit semantic to ETX => typed options isn't possible? (as well as me better understanding... are we talking about something like settings stored in a set order and assuming that order won't change, vs say in a json list?

raphaelcoeffic, 2021-06-18:
> So another known limitation for the notes? i.e.
>
> * Panel / widget settings cannot be preserved in the conversion from OTX 2.3.x => ETX 2.4.0 due to incompatibility in how the settings sere stored. [...]

2.3 does not include the type of the options in the binary file.

pfeerick, 2021-11-28:
This should be resolved between access builds being made available and https://github.com/EdgeTX/edgetx/pull/991 adding ACCESS menus for 2.6.

closed 2021-11-28 by pfeerick