EdgeTX/edgetx #247: "About EdgeTX" in main menu needs content

rotorman, 2021-06-17:
Before final release of v2.4.0, it would be nice to have some text pop-up to be shown to user when clicking "About EdgeTX" in the main menu.

raphaelcoeffic, 2021-06-18:
@rotorman whereby this has not really been a high priority to me, it is definitely needed. Anyone up to drafting something?

rotorman, 2021-06-18:
Maybe a native speaker with witty tongue could make a proposal. @pfeerick I am winking towards you ;)))

raphaelcoeffic, 2021-06-18:
> Maybe a native speaker with a witty tongue could make a proposal. @pfeerick I am winking towards you ;)))

So do I ;-)

pfeerick, 2021-06-18:
Oh wait, you just mean the text? I can do that... thought you'd want to me make it actually work also... 😆 

That's easy... depending on space. 

```
EdgeTX 2.4.0 - "Mostly stable"
Copyright © 2021 EdgeTX

Website: https://edge-tx.org/
GitHub: https://github.com/EdgeTX/edgetx

Based on code named
  - opentx - http://github.com/opentx/opentx
  - th9x - http://code.google.com/p/th9x
  - er9x - http://code.google.com/p/er9x
  - gruvin9x - http://code.google.com/p/gruvin9x

License GPLv2: http://www.gnu.org/licenses/gpl-2.0.html

This program is free software; you can redistribute it and/or modify
it under the terms of the GNU General Public License version 2 as
published by the Free Software Foundation.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU General Public License for more details.
```

Maybe change that first line to
```
EdgeTX 2.4.0 - "Are We There Yet?"
```
a.k.a. the milestone name ;)

raphaelcoeffic, 2021-06-18:
i love the idea of having release names, rather than just numbers.

raphaelcoeffic, 2021-06-18:
And of course the logo :-) We could even scrap most of it and add a QR code that redirects to a page with these info.

raphaelcoeffic, 2021-06-18:
so like:
```
EdgeTX 2.4.0 - "Mostly stable"
Copyright © 2021 EdgeTX

[QR code]
```

pfeerick, 2021-06-18:
I would do a QR code in conjunction with the text... or at absolute minimum do:

```
EdgeTX 2.4.0 - "Are We There Yet?"
Copyright © 2021 EdgeTX

Website: https://edge-tx.org/
GitHub: https://github.com/EdgeTX/edgetx
License:  GPLv2

[QR Code - Welcome Page with rest of the details]
```

rotorman, 2021-06-18:
How about the following for the release names:
2.4.0 - "The breakthrough" (I would prefer the name to radiate positiveness, which is why mostly stable and are we there yet IMHO would be less nice)
2.4.1 - "The dark edition"
2.4.2 - "Santa's Christmas Special"

pfeerick, 2021-06-18:
You can see the current milestone names here: https://github.com/EdgeTX/edgetx/milestones

I have no objections to the proposed changes... especially since I knew no-would would link "mostly stable" to "mostly harmless" 😆

raphaelcoeffic, 2021-06-18:
We need to hide "42" somewhere ;-)

pfeerick, 2021-06-18:
Don't forget ... *"Never leave home without a towel"*

lshems, 2021-06-18:
Normally release names start with subsequent letters, and are meant to be catchy to remember. The phrase after the namecan be anything creative to exoress the feeling of the moment.

How about:

Aligned, a new start
Bold, some wild ideas
Cubicle, new things out of the box
Diagonal, the shortest path between opposite corners
E

And the 42 should be engraved inside the precious

lshems, 2021-06-18:
They are all geometric terms, linked to 'edge'

JimB40, 2021-06-20:
So let’s go with more edges with evey major.
Triangle, square, pentagon, hexagon …
Easy to visualize and names are not copyrighted.
Until we reach Decagon RC models will be mind-controlled anyway :)

lshems, 2021-06-20:
Order them alphabetical, and leave the current proposed milestone names as description, passphrase, however you call it.
But it's not up to me. Just suggestions

NemesisXB, 2021-06-20:
> Order them alphabetical, and leave the current proposed milestone names as description, passphrase, however you call it.
> But it's not up to me. Just suggestions

I agree it should be alphabetical. Why not aircraft names?

lshems, 2021-06-20:
I like the geometrical link to edge as an application name.

NemesisXB, 2021-06-20:
> I like the geometrical link to edge as an application name.

I understand the link. But that is do boring. Just my 2c.

lshems, 2021-06-20:
Boring is very subjective. 

I can provide a plane manufacturing list alphabetical. I proposed it for releases of an app on my work. We got quite far in the alphabet.

rotorman, 2021-06-23:
Solved by efc947d
[image]

closed 2021-06-23 by rotorman