EdgeTX/edgetx #248: Model Name missing from selection window

grepboy, 2021-06-17:
When going to the model selection menu, some model names are not printed on the main selection list.  You can see the model name by selecting the model then going into Edit Model and the name will show there.

Seems some names will display and some will not.  

Please view video at 
[https://youtu.be/5271M3XdI-0](https://youtu.be/5271M3XdI-0)

pfeerick, 2021-06-17:
Any idea what the common thread between the names being shown and not shown? What happens if you change a character in a name, and then change it back again?

eshifri, 2021-06-17:
If the name is edited in edgeTX, it is being displayed correctly.
In my case, models edited in 2.3.11 were seen from the very beginning; models edited in the older OTX versions required this edit->change->restore exercise.

grepboy, 2021-06-18:
Confirmed. Editing model name does make it display in model select menu.

raphaelcoeffic, 2021-06-18:
@grepboy so it seems like a conversion error, right?

pfeerick, 2021-06-18:
> @grepboy so it seems like a conversion error, right?

It looks like it, and related to pre 2.3.11 OTX models

raphaelcoeffic, 2021-06-18:
@grepboy Could you please upload the RADIO & MODELS directories (ZIP file at best)? We would need the files **before** the conversion, so I can test the conversion itself.

lshems, 2021-06-18:
Where is the conversion done exactly. what is the conversion procedure?

My models loose way more than just the model name, but I'm just copying the bin files over the the new sdcardincluding the models.txt

grepboy, 2021-06-18:
[OTX-PRECONV.zip](https://github.com/EdgeTX/edgetx/files/6677853/OTX-PRECONV.zip)
I've uploaded the pre-converted models and radios.  This is from OTX 2.3.11

grepboy, 2021-06-18:
Sorry, I also forgot to say - I did use OTX Compantion to load these models into the new structure. 
[tx16s.zip](https://github.com/EdgeTX/edgetx/files/6677862/tx16s.zip)
I've attached  my saved models from there as well.

commit 82f60cdc (2021-06-19): Fix model names on model select page with not yet converted models

Resolves #248
closed 2021-06-19 by raphaelcoeffic