EdgeTX/edgetx #251: "Time" does not show anything on widget

eshifri, 2021-06-17:
**Describe the bug**
When "Time" is selected for widget, the widget shows the word "time", but not ant value.

 - EdgeTX version  2.4.0 current code compiled.
 - Hardware RM TX12S

pfeerick, 2021-06-17:
A `Value` widget with the source set to `Time`? Confirmed on RM TX16S, 2.4.0 RC1

eshifri, 2021-06-18:
Yes, Value widget and RM T16S (not TX12)

eshifri, 2021-06-18:
File /widgets/value.cpp contains Timer field and does not have Time field.

eshifri, 2021-06-18:
The following addition to values.cpp around line 93 fixis it:

      if (field == MIXSRC_TX_TIME) {
        drawTimer(dc, xValue, yValue, getValue(MIXSRC_TX_TIME),
                  attrValue | FONT(XL) | CUSTOM_COLOR);
      }

closed 2021-06-19 by raphaelcoeffic