EdgeTX/edgetx #252: Timer widget does not allow to change font color

eshifri, 2021-06-18:
**Describe the bug**
When Timer widget is created there is no way to change the font color.
Also the font size is smaller than for other widgets.

**General information**
 - EdgeTX  2.4.0 current code compiled
 - Hardware RM TX16S

eshifri, 2021-06-18:
timr.cpp has the following code 
const ZoneOption TimerWidget::options[] = {  { STR_TIMER_SOURCE, ZoneOption::Timer, OPTION_VALUE_UNSIGNED(0) },  { nullptr, ZoneOption::Bool }};

and does not have the color and shadow selection:

{ STR_COLOR, ZoneOption::Color, OPTION_VALUE_UNSIGNED(WHITE) },  
{ STR_SHADOW, ZoneOption::Bool, OPTION_VALUE_BOOL(false)  }

pfeerick, 2021-06-18:
hawkeyed on Discord noted that:
> eshri has a report on a timer with no colour adjustment, if he is talking about timer 1 in top left tool bar then neither does OpenTx but Timer 2 the large font version does have colour adjustment

However, it if you wanted it as a top bar widget, that won't help you as it doesn't seem to work/show when on the top bar.

eshifri, 2021-06-18:
This is not in the tool-bar - I'm talking about any Timer1 widget on any telemetry screen.
The timer widget configuration does not have color selection and shadow options.
This would not be a big deal, but the timer font is white, so on the bluish background it is not well visible.
timer.cpp code does not support color changes, if I read it correctly.

pfeerick, 2021-06-18:
You'll need to be more specific (for me, anyway!!)

This screenshot shows Timer (top left), Timer2 (bottom left), Value - Tmr1 (top right). Both Timer2 and Value let you set the colour. 
[image]

Timer doesn't normally *need* to be able to set text colour when it's in a normal panel, because it has it's own background, but when you go to smaller layouts like the 4x2 layout, or use it on the top bar, it loses the background, so at that point it does (have a need to be able to set the colour). Is this the sort of situation you mean?

raphaelcoeffic, 2021-06-18:
@pfeerick looking at the regular "Timer" widget, it seems it is still using the wrong color, by the way ;-) I would use the darker blue.

eshifri, 2021-06-18:
Oh, looks like I need to use VALUE->Timer - not timer.
What is the difference between the two?
And I was talking about the 2x4 widget layout. In this case Value->Timer1 has normal color control, but Timer1 - does not.

pfeerick, 2021-11-28:
What do you think @eshifri ... is this still needed since build in Timer now follows theme colours?

eshifri, 2021-11-28:
I think it can be closed.

closed 2021-11-28 by pfeerick